slipx06/sunsynk-power-flow-card #489: Add Time Of Use SOC to shut off the battery

SirUlbrich, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently the card doesn't show the time of use SOC, you can just put inside off grid and shutoff soc of the inverter. not the shut off battery by time of use (deye / sunsynk) 

### Expected behaviour

by use power out of the battery showing the soc by time of use with a template sensor like:

`      state: >
        {% if states("sensor.time") >= states("select.esp32_sun12k_tou1_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou2_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_1_capacity") }}
        {% elif states("sensor.time") >= states("select.esp32_sun12k_tou2_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou3_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_2_capacity") }}
        {% elif states("sensor.time") >= states("select.esp32_sun12k_tou3_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou4_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_3_capacity") }}
        {% elif states("sensor.time") >= states("select.esp32_sun12k_tou4_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou5_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_4_capacity") }}
        {% elif states("sensor.time") >= states("select.esp32_sun12k_tou5_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou6_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_5_capacity") }}
        {% elif states("sensor.time") >= states("select.esp32_sun12k_tou6_startzeit") and states("sensor.time") <= states("select.esp32_sun12k_tou1_startzeit") %}
          {{ states("number.esp32deyedsm_sun12k_time_point_6_capacity") }}
        {% else  %} {{ "n.V." }}
        {% endif %}`

### Possible Solutions

[image]

possible to use shutdown or time of use SOC - locked against each or something else. if you use time of use you cant go deep in the shutdown / low batt.

### Mode

Both

### Context / Reason

When using the time of use, this SOC is the relevant one to switch off the battery or no more power is used from the battery

slipx06, 2024-07-08:
Hi Not sure I understand this request. The card does display the time of use SOC but you need to define the sensors below. The SOC that you have highlighted in red will adjust based on the time of use schedule

```
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_prog1_capacity
  prog1_charge: switch.sunsynk_prog1_grid_charge
  prog2_time: sensor.sunsynk_time_slot_2
  prog2_capacity: number.sunsynk_prog2_capacity
  prog2_charge: switch.sunsynk_prog2_grid_charge
  prog3_time: sensor.sunsynk_time_slot_3
  prog3_capacity: number.sunsynk_prog3_capacity
  prog3_charge: switch.sunsynk_prog3_grid_charge
  prog4_time: sensor.sunsynk_time_slot_4
  prog4_capacity: number.sunsynk_prog4_capacity
  prog4_charge: switch.sunsynk_prog4_grid_charge
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.sunsynk_prog5_capacity
  prog5_charge: switch.sunsynk_prog5_grid_charge
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.sunsynk_prog6_capacity
  prog6_charge: switch.sunsynk_prog6_grid_charge
```

SirUlbrich, 2024-07-08:
Yeah. That's what I mean. You can delete the request. Sorry. My fault.

closed 2024-07-08 by slipx06