slipx06/sunsynk-power-flow-card #493: Deye 20-50KW HV two Batteries

Mights001, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hello guys, I have a 50KW Deye Hybrid, the inverter has 2 battery connections. Would it be possible to expand your dashboard for two batteries?

[image]
[image]

Thanks in Adv.

### Expected behaviour

.

### Mode

lite

### Context / Reason

Lite Mode i prefer

slipx06, 2024-07-26:
Hi. This has been requested a few time. I haven't had a chance to look at it yet but will gladly accept pull requests :)

JanKraslice, 2024-11-13:
I am also asking very much for the addition of a second battery, I will financially support the development.
I have an identical DEYE 30Kw HV inverter and two batteries connected

I am adding a download request and will be happy to financially support and contribute to your project

PR #561 (2024-11-25): Add a second battery

## Description

Adds the ability to display a second battery when using the wide screen option on all card styles
## Related issues

Closes #493
Also requested and discussed here #380 #374 #296 #285 #232 #203 #135
## Motivation and Context

Multiple requests to add a second battery. 
## How has this been tested

I've tried to cover all scenarios but I am not able to test this fully. 
## Type of change
- New feature (non-breaking change which adds functionality)