slipx06/sunsynk-power-flow-card #495: PV Efficiency - store todays max value or enable statistics on the % field.

KajLehtinen, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When using the pvX_max_power field to enable displaying PV efficiency (PV1-4) the current % efficiency is shown, but I think that its more useful to see the daily top value to know how the paneles has performed during the day. Can this GUI field be enabled for statistics so its easy to review the panels performance after the fact.

/Kaj 

### Expected behaviour

Collect the daily statistics as it happens and store.

### Mode

full

### Context / Reason

I would find it useful to see the statistics for the whole day of the PV efficiency in percent - preferably if possible, for the whole year.

slipx06, 2024-07-26:
Hi. Those variables are calculated using the card's internal logic and not stored in the HA DB. As such there is no statistical data available. You would need to create a template sensor within HA that performs this calculation if you want saved state and statistics. This sensor could then be displayed on the card.

KajLehtinen, 2024-07-28:
OK I understand, but is it possible to add a option show todays highest value instead of current % value? This way one can atleast see how efficient the panels has been over course of the day without looking at the display when it happens.

slipx06, 2024-07-28:
You can also just click each of the PV values and it will bring up the more info dialog which will show you this info
[image]