slipx06/sunsynk-power-flow-card #500: Card broken in Homeassistant App and some android devices since latest Homeassistant Update

jzielke84, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Card looks like this:
[image]

### Expected behaviour

Correct Style

### Card Version

4.44.0

### Home Assistant Version

2024.8.0

amurray81, 2024-08-09:
Also see same. The full card still works if you need a temporary fix

slipx06, 2024-08-09:
I cant reproduce this one yet. Tested on ipad, android mobile and windows. Chrome and companion app with latest 2024.8.0. Is anyone else seeing this?

freddanastrom, 2024-08-09:
Same for me after updating to 2024.8
Card is rendered like the picture above on my iPad, but iPhone app on the other hand works fine.

amurray81, 2024-08-09:
[image]
[image]

App and tablet at same time.. Excuse the camera shot but it's a pain to sign into

watersnake154, 2024-08-09:
Same problem on the iPhone everything as always and on the tablet the identical error.
[image]

jzielke84, 2024-08-09:
Here are my results:

- MacOS Chrome ✅ 
- MacOS Safari ✅ 
- iOS Safari ✅ 
- iOS Chrome ✅ 
- Homeassistant App MacOS ❌ 
- Huawei Matepad 7 ❌

sitvesep, 2024-08-09:
It happened to me on an iPad where I updated iOS to 17.6.1. Web interface ok, app error. Interestingly, the error is only with card style litle, if I switch it to full or compact it is OK

pepe895, 2024-08-10:
I have same problem at Galaxy Tab with Kiosk 6.0.1. Only full style working.

jzielke84, 2024-08-10:
Another custom card also shows some strange values „xx px“ with that new version. So I guess there was a breaking change in HA and some cards have to be updated. The channelog however is huge.

slipx06, 2024-08-10:
@jzielke84  Are you seeing any errors in the logs? Whats other custom cards are not working for you? 
I've also tested FireFox on Windows with no Issues
@Pho3niX90 do you have a Mac to check on your end?

jzielke84, 2024-08-10:
> Are you seeing any errors in the logs?
Unfortunately the HA companion app neither the browser in my tablet (Fully Kiosk) offers a debug mode like Chrome.

> Whats other custom cards are not working for you?
https://github.com/Yevgenium/weather-chart-card

slipx06, 2024-08-10:
OK seems to be a bigger issue affecting other cards 

https://github.com/home-assistant/iOS/issues/2908

parkflyer, 2024-08-18:
As the „bigger issue“ is closed now (without any fixes), how can this be solved? The issue is present on my iPad too…

jzielke84, 2024-08-18:
@parkflyer all custom cards using dynamic CSS have to do change their code.

slipx06, 2024-08-20:
Hi All, The card makes extensive use of dynamic css and styles and so the work around here is to either: 

- Stay on 2024.7.4 
- Upgrade your device/browser 
- Change/spoof your user agent string

jzielke84, 2024-08-20:
> * Stay on 2024.7.4
Out of the question

> * Upgrade your device/browser
Would like to but can't do it on my Tablet

> * Change/spoof your user agent string
Tried but didn't solve the problem.

So I guess I'l gave to go back to power-flow-card-plus which does the job...

closed 2024-08-20 by slipx06

parkflyer, 2024-08-20:
Please note that I am not talking about any browser here, I'm using the Home Assistant app on my iPad.
So also for me your options are not really helping. 
Or should we contact the guys for the HA iPadOS app? Because on the iPad in the Safari browser there is no problem, just in the app...

jzielke84, 2024-08-20:
@parkflyer Well Homeassistant is not responsible for 3rd party cards. So if they change the codebase it's up to the maintainers of the corresponding card to fix it. Hence they probably won't change anything on their companion app, since it runs fine with the stock cards.

pepe895, 2024-08-21:
I tried update my Galaxy Tab to Android 11, but card have same problem. Problem is strange because the card always works for a while. Sometimes an hour, sometimes a whole day.

slipx06, 2024-08-21:
> Please note that I am not talking about any browser here, I'm using the Home Assistant app on my iPad. So also for me your options are not really helping. Or should we contact the guys for the HA iPadOS app? Because on the iPad in the Safari browser there is no problem, just in the app...

There was an issue with the companion app on the ipad that has already been addressed. So hopefully this is fixed in the next HA release

sitvesep, 2024-08-22:
An update to the iOS app was released today. It's working properly on my iPad again.

parkflyer, 2024-08-22:
As assumed a fix in the iPad app was necessary. Works for mee too.