slipx06/sunsynk-power-flow-card #509: card is not displayed when running in android.webview

norbi771, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When sunsynk-power-flow-card is running in android webview the card is not displayed, and error message says:
Custom Element doesn't exist: sunsynk-power-flow-card

[image]

Webview Provider: com.android.webview
Webview Version: 83.0.4103.120

I tried to force cache update, suspecting that there might be issue with old cached js. So I deleted cache and storage in android way, changed name of JS files in resources and filesystem in HA order to force the JS to reload. It didn't help

### Steps To Reproduce

1. This issue occurs on tablets with android 11. 
2. Problem shows up, when using the home assistant application. It does not seem to exist when using pure chrome.
2a. Problem could be also reproduced when you connect to home assistant address from kiosk software, that is using webview component,  I tried fully browser, https://www.fully-kiosk.com/files/2024/07/Fully-Kiosk-Browser-v1.56.1.apk

[image]

I caputed this log with web console from fully browser (browser_mod mentioned in the screenshot was removed and it does not make a change, the problem was confirmed also on freshly started identical tablet)
[image]

### Expected behaviour

Display power flow card like in Chrome
[image]

### Card Version

4.43.0

### Home Assistant Version

2024.8.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
width: 100%
large_font: true
decimal_places: 2
dynamic_line_width: true
max_line_width: 8
show_battery: true
panel_mode: false
show_grid: true
battery:
  energy: 20480
  shutdown_soc: 20
  show_daily: true
  show_remaining_energy: true
  auto_scale: true
  show_absolute: true
  invert_power: false
  colour:
    - 0
    - 86
    - 214
  dynamic_colour: true
  linear_gradient: true
solar:
  show_daily: true
[... 685 lines cut ...]
    getUpdateComplete() {
        return this._$ES
    }
    shouldUpdate(e) {
        return !0
    }
    update(e) {
        this._$Ej &&= this._$Ej.forEach((e => this._$EC(e, this[e]))),
        this._$EU()
    }
    updated(e) {}
    firstUpdated(e) {}
}
;
_.elementStyles = [],
_.shadowRootOptions = {
    mode: "open"
},
_[m("elementProperties")] = new Map,
_[m("finalized")] = new Map,
f?.({
    ReactiveElement: _
}),
(p.reactiveElementVersions ??= []).push("2.0.4");
```

slipx06, 2024-09-02:
Hi. Looks like the issue is related to https://github.com/slipx06/sunsynk-power-flow-card/issues/500 and the fact that you are using what HA considers to be a legacy browser. 

See also

https://github.com/home-assistant/iOS/issues/2908
https://github.com/home-assistant/frontend/issues/21004

norbi771, 2024-09-03:
Hi, 
Thank you for prompt answer.
Indeed, there are issues. Although I spend some time searching I didn't see them earlier.
I think I will wait for HA update and see if anything has changed.
Thank you,

BTW, the sunsynk-power-flow-card is really nice :-)
Keep up the good work, thank you!

closed 2024-09-05 by slipx06