slipx06/sunsynk-power-flow-card #512: Strange corruption on macOS Safari

quentinsf, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Anybody seen strange corruption like this on Safari?

[image]

It seems to be a problem parsing some of the SVG - I'm getting console errors like:

`Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"`

It works fine on other browsers on the same machine, and in the HA app, and in Safari on my iOS devices.

However, I *do* see the same thing on two different Macs, running slightly different versions of Safari and very different versions of macOS. So it's not just my one individual machine, and I've tried flushing caches, reloading resources etc.

This started when I updated Home Assistant to 2024.9.1 and the Sunsynk card to v4.44.0 at the same time, so I'm afraid I can't tell which was responsible.  My guess is that the SVG parser on macOS Safari may be fussier than some others, and there's a closed quote missing or something.

It may, of course, be something about my config, so I'm curious to know if others are running the latest version successfully on macOS Safari?

### Steps To Reproduce

e.g. in Safari 16.6 on macOS Ventura, load a page with the Sunsynk card

### Expected behaviour

I've been seeing a lovely card up until this latest update.

### Card Version

v4.44.0

### Home Assistant Version

2024.9.1

### Relevant log output

```shell
[Error] Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Problem parsing d="3.37875"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Problem parsing d="96"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Problem parsing d="3.380625"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Problem parsing d="94"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
[Error] Error: Invalid value for <svg> attribute y="t=>xn.handlePopup(t,e.entities.grid_connected_status_194)"
	j (sunsynk-power-flow-card.js:24:5992)
	_$AI (sunsynk-power-flow-card.js:24:5911)
	p (sunsynk-power-flow-card.js:24:3620)
	$ (sunsynk-power-flow-card.js:24:4679)
	_$AI (sunsynk-power-flow-card.js:24:4176)
	(anonymous function) (app.wNoxo8Q30IA.js:2:352642)
	(anonymous function) (sunsynk-power-flow-card.js:30:512)
	update (sunsynk-power-flow-card.js:30:519)
	performUpdate (sunsynk-power-flow-card.js:12:5142)
	(anonymous function) (sunsynk-power-flow-card.js:12:4594)
```

itsgucci, 2024-09-14:
I am experiencing the same issue of the icons not showing up, but mine is running in the macos Home Assistant app and not through a browser. 
[image]

It does work if I go into edit the cards, and select lite version. (I am using the full) until I quit and relaunch the app, then the lite will be corrupted, and I can select the full, which works until I quit the app. 
[image]
[image]
[image]

So it seems it is there when first selected, but after relaunching something fails.

[image]

macos sonoma 14.6.1
card v4.44.0

slipx06, 2024-09-15:
Hi

Unfortunately a know issue with dynamic css and styles and what HA considers to be a legacy browser. See https://github.com/slipx06/sunsynk-power-flow-card/issues/500

You can revert back to 2024.7.4 or upgrade your device/browser

quentinsf, 2024-09-15:
> Unfortunately a know issue with dynamic css and styles and what HA considers to be a legacy browser. See #500

Ah, but my Safari is the latest version and way newer than anything labelled ‘legacy’ there.

pepe895, 2024-09-20:
Its not about system or browser. A have same problem at tablets with Android 11 + 12 and new browser. I also have old phone with Android 9 where card working without problem.

purewhitedesign, 2024-09-21:
Same here. A few days ago it looks like the posted screenshots above. But only on the iPad Air 4th. On PC and iPhone it works flawless!

iPhone:
[image]

iPad:
[image]

purewhitedesign, 2024-09-21:
Problem solved....(in my case)
Update the home assistant app in the app store.

slipx06, 2024-10-28:
See https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v5.0.0

Linking to other reported problems 
#529 
#511 
#509
#500

closed 2024-10-28 by slipx06