slipx06/sunsynk-power-flow-card #515: Add car battery %

Ironrampant, 2024-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]

In my example picture the power to the car is replicated in 2 locations (7.1 kW in this case). The "non-essential" text below is also wasted space, as the car icon shows what it is.

### Expected behaviour

I'd like to replace something with the car battery %

### Possible Solutions

 Maybe an option to use the lower box if there is only one non-essential?  Or maybe swapping out the non-essential text field?

### Mode

lite

### Context / Reason

Utilises space better and reduces the need for having the car battery percent on other parts of the screen

slipx06, 2024-09-29:
Hi. You could set `additional_loads: 0` to remove the duplicate box assuming there is nothing else connected to the non-essential circuit and set the non essential icon; `nonessential_icon: mdi:car`. You can also remove the Non Essential name or set is to something else; `nonessential_name: " "` 

[image]

Alternatively you can use the `non_essential_load1_extra: <battery_sesnor>` to show additional information. You can still remove the non essential name to free up space but you will have the duplicate data if nothing else is connect to your non essential circuit

[image]

The final option is to use the AUX input to show this

[image]

Have a look at these sensors
```
aux_name: Car
aux_daily_name: DAILY CAR
aux_type: mdi:car
aux_power_166: input_number.test_number
aux_load1_extra: sensor.sunsynk_battery_capacity_shutdown
```

slipx06, 2024-09-29:
Also looks like you are using an old version of the card. You should update it as some things have changed