slipx06/sunsynk-power-flow-card #517: Have separate sensors for battery charge/discharge

pameruoso, 2024-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I understand that the card shows the power battery discharge and charge direction and power based on the sign of the value passed by the sensor (must provide one sensor)

### Expected behaviour

 With growatt  you have 2 separate sensors both positive. if battery is discharging the specific sensor is positive and shows how much power the battery is giving to the load. If the battery is charging the discharging sensors is 0 and the charging one is higher than 0.

### Possible Solutions

 Is it possible to have 2 separate sensors that show the battery power usage and hence the power direction from the battery?

### Mode

Both

### Context / Reason

adding capability for growatt inverter

slipx06, 2024-10-04:
Hi. Quick fix is to just create a template sensor. 

```
template:
  - sensor:
      - name: "Battery Power"
        unique_id: battery_power
        device_class: "power"
        unit_of_measurement: "W" 
        icon: mdi:battery
        state_class: measurement
        state: >
          {% set charge = states('sensor.battery_charge') | float %}
          {% set discharge = states('sensor.battery_discharge') | float %}
          {{ (discharge - charge) |float(0)|round(0) }}
```

Grosi001, 2024-10-10:
Hi
I have a similar problem, with the invert power function. I tried to make a template for this boolean value:

`battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
  show_absolute: false
  invert_power: sensor.battery_charging_status
  auto_scale: true
  dynamic_colour: true
  animate: false
  hide_soc: true`
 
 But that doesn't work. Any suggestions?

slipx06, 2024-10-14:
@Grosi001 not sure I understand. What template sensor are you trying to create?

Grosi001, 2024-10-15:
> @Grosi001 not sure I understand. What template sensor are you trying to create?

Sorry if I expressed myself confusingly. I want the direction of flow to change when the battery is charging or discharging. Since the card does not recognize this itself, I have created a template that changes from true to false depending on the state (charging or discharging). Then I wanted to use this template with invert_power so that the flow direction changes. 

Maybe important: I'm using a inverter from Sungrow.

slipx06, 2024-10-17:
Hi. The direction is determined by the battery power. If it is either positive or negative. You should build you template sensor to adjust this value rather than try dynamically change the invert parameter.

slipx06, 2024-10-28:
I've added another battery attribute `invert_flow`. It may not solve your problem but adds more flexibility.