slipx06/sunsynk-power-flow-card #519: Essential shows wrong value

papperone, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

My Grid sensor is opposite (negative for import / positive for export) so I setup "invert_grid: true" and this is now showing correctly in the Grid box, but the Essential calculation still consider the wrong value (negative) so it looks like my house is feeding power to inverter which is obviously impossible due to the wrong calculation as explained (see below picture); I think that when I select "invert_grid: true" the formula to calculate essential should consider the opposite value of the sensor so it looks to me like a bug
[image]

### Expected behaviour

Expect that when I use "invert_grid: true" the value in the Essential formula will be considered accordingly but it's not

### Card Version

5.0.0

### Home Assistant Version

2024.9.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: false
large_font: false
title: Fotovoltaico
title_colour: grey
title_size: 24px
show_solar: true
show_battery: true
show_grid: true
decimal_places: 2
dynamic_line_width: true
min_line_width: 1
max_line_width: 4
inverter:
  modern: true
solar:
  mppts: 2
  show_daily: true
  max_power: 5900
battery:
  energy: 20000
  shutdown_soc: 10
  show_daily: true
  invert_power: false
  colour: green
  animation_speed: 6
  max_power: 5000
  show_absolute: false
  auto_scale: true
  hide_soc: false
  dynamic_colour: true
  linear_gradient: true
  animate: true
  path_threshold: 100
load:
  show_daily: true
grid:
  show_daily_buy: true
  show_nonessential: false
  invert_grid: true
entities:
  inverter_voltage_154: sensor.ftv_r_phase_inverter_voltage
  load_frequency_192: sensor.ftv_r_phase_inverter_frequency
  inverter_current_164: sensor.ftv_r_phase_inverter_current
  inverter_power_175: sensor.ftv_r_phase_inverter_power
  grid_power_169: sensor.ftv_total_grid_power
  battery_voltage_183: sensor.ftv_battery_voltage
  battery_soc_184: sensor.ftv_battery_soc
  battery_power_190: sensor.ftv_battery_power
  battery_current_191: sensor.ftv_battery_current
  grid_ct_power_172: sensor.ftv_r_phase_grid_power
  inverter_status_59: sensor.ftv_inverter_status
  pv1_power_186: sensor.ftv_pv1_power
  pv2_power_187: sensor.ftv_pv2_power
  pv1_voltage_109: sensor.ftv_pv1_voltage
  pv1_current_110: sensor.ftv_pv1_current
  pv2_voltage_111: sensor.ftv_pv2_voltage
  pv2_current_112: sensor.ftv_pv2_current
  day_pv_energy_108: sensor.ftv_daily_production
  day_battery_charge_70: sensor.ftv_battery_daily_charge
  day_battery_discharge_71: sensor.ftv_battery_daily_discharge
  day_load_energy_84: sensor.ftv_daily_consumption
  day_grid_import_76: sensor.ftv_daily_energy_purchased
```

papperone, 2024-10-08:
I may not have explained correctly, both grid_ct_power_172 and grid_power_169 in my inverter expose the same value and both are opposite sign (negative I'm importing from grid, posivite I'm exporting); so I'm expecting that with the parameter "invert_grid: true" both will change sign showing the right value, in the picture above the flow is from the house to the inverter of more than 2000Wh while in reality it's the other way around, my house is consuming that amount that shoudl be then considered as positive (because of the invert parameter) and then have the flow of energy in the other direction.... if this is not the case I may miss something to have the card showing the data of my system correctly...

look now, it's night, battery are fully used (below 10%) so my house shoudl show positive consumption all from grid but it shows negative like it's feeding the grid :( how can I fix this behaviour ??

[image]

slipx06, 2024-10-09:
So those are two different measurement points on the Sunsynk. A quick solution is to create a template sensor that inverts your sensor value and then you can set` invert_grid: false again`

```
template:
  - sensor:
      - name: "ftv_total_grid_power_inverted"
        unique_id: total_grid_power_inverted
        device_class: "power"
        unit_of_measurement: "W" 
        state_class: measurement
        state: >
          {{ states('sensor.ftv_total_grid_power') |float(0) *-1 }}
```

papperone, 2024-10-09:
> So those are two different measurement points on the Sunsynk. A quick solution is to create a template sensor that inverts your sensor value and then you can set` invert_grid: false again`
>
> ``` [...]

thanks, I thought as well about this workaround, I thought it coudl have been implemented relatively easier in the card but I  will go for it ;)

closed 2024-10-12 by slipx06