slipx06/sunsynk-power-flow-card #521: Interactive icons

Sn0w82, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently the Sun/Invertor/Battery/Grid/Essential/Non-Essential icons are not interactive or cannot be set as interactive.

### Expected behaviour

Give the user the possibility to make them interactive and be able to link to them subviews or other dashboards/tabs

### Mode

Both

### Context / Reason

Usage improvement and coverage of both executive summary that cards gives AND fast dive deeper in details for a specific component if user wants this.

slipx06, 2024-10-14:
Thanks for the suggestion, Will take a look

closed 2024-10-16 by slipx06

slipx06, 2024-10-20:
@drtinaz this has been added to v5.2.0

Sn0w82, 2024-10-20:
> @drtinaz this has been added to v5.2.0

Hey slip,

you added for all icons/sections except grid icon/section. In 5.2.0 there is no navigation section available to setup the link under grid section.

slipx06, 2024-10-20:
@Sn0w82 Its there. Make sure you reset your front end cache

Sn0w82, 2024-10-20:
I did reset the frontend after initial update and i did now again. Same, i have all other entities navigation tabs but not for grid
[image]

slipx06, 2024-10-20:
It must still be using the old cached card. 

If you using a browser push F12 and then long press the reload button

[image]

If you are using the IOS companion app. Side bar> app configuration > debugging>reset frontend cache

Sn0w82, 2024-10-20:
I saw you just pushed an  update. Now grid have navigation tab too and works when pressed. 

Thanks.

slipx06, 2024-10-20:
It was just an update to the documentation. Anyway glad it’s working.

commit b37479c2 (2024-10-16): feat: you can specify the battery charge cut-off capacity using the `soc_end_of_charge` attribute. If for example you set your battery to charge up to 90% the card will correctly display charge time to this capacity. Expects a numeric value between `80` and `100` or sensor i.e. `sensor.soc_end_of_charge`. closes #514
feat: introduces the `navigate` attribute that allows you to link to other dashboards and views e.g. `/lovelace/1` when clicking some of the card icons. It can currently be configured for the battery icon, essential load icon, solar icon, and inverter icon. closes #521