slipx06/sunsynk-power-flow-card #522: The TOD SOC display is not correct when prog6 time is earlier than others.

stealth-eye, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The battery area shows two SOC when TOD settings are enabled, current soc and tod soc. The TOD soc is not displayed correctly for most of the day when prog6 is earlier than rest.

[image]

My tod settings: (time in 24hrs)
prog1_time = 11:00
prog2_time = 12:00
prog3_time = 18:00
prog4_time = 22:00
prog5_time = 1:00
prog6_time = 8:00

SOC
prog1_capacity = 65
prog2_capacity = 75
prog3_capacity = 60
prog4_capacity = 50
prog5_capacity = 35
prog6_capacity = 40

During daytime after 8AM, it almost always shows the prog6 capacity instead of current time segment's capacity (as shown in the screenshot, its showing 40%. Screenshot is taken around 4.15PM).

### Expected behaviour

TOD SOC should be shown correctly based on current time.

### Card Version

5.0.0

### Home Assistant Version

2024.10.1

closed 2024-10-14 by slipx06

slipx06, 2024-10-14:
@stealth-eye please test 5.0.1

stealth-eye, 2024-10-14:
Tested on 5.0.1, Works fine.
Thank you.

slipx06, 2024-10-14:
Thank you for the coffee 😄

stealth-eye, 2024-10-14:
The pleasure is all mine. Sincerely grateful for this widget and its documentation. Helped me setup a local dashboard instead of relying on vendor's mobile app.

stealth-eye, 2025-04-05:
I would like to report a regression on this issue. From 12AM, It is showing "Shutdown SOC" value in the "Runtime to" instead of configured next % slab. 

[image]

slipx06, 2025-04-05:
Hi. This is by design. When SOC = current program capacity it will show the runtime to shutdown SOC.

commit ac3a52c6 (2024-10-14): fix: better handling of system timer closes #522