slipx06/sunsynk-power-flow-card #526: Latest version  5.1.1 - Daily Sell does not reset

wgumaa, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The daily sell did not reset to 0 after upgrade to the latest version.  

[image]

### Expected behaviour

Daily sell should go back to 0 just after midnight

### Card Version

5.1.1

### Home Assistant Version

2024.10.2

wgumaa, 2024-10-18:
Also the Daily Grid Buy did not reset too.

wgumaa, 2024-10-18:
It could be a problem from my smart meter

slipx06, 2024-10-18:
Hi. You will need to check your sensor data. The card just displays the sensor value and will report what it receives

wgumaa, 2024-10-18:
For some reason, my sensors did not reset and I had to recreate them. Its a sunny day today so I am sending to the grid so at the moment my grid consumed sensor is showing as unknown as there is no electricity coming in at the moment. This is causing my daily load template sensor to show unknown as well.

closed 2024-10-18 by wgumaa