slipx06/sunsynk-power-flow-card #528: Please add V2H Feature to card

terzo33, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I can't see V2H (vehicle to home) icon and sensors (hz,V,A). Actually I'm used Grid Power for V2H but there is no way to see how much power the two power inputs are sending to the inverter. the only possibility at the moment is to choose V2H or grid power :(

### Expected behaviour

Please add V2H (vehicle to home) icon and sensors (Hz,V,A) near Grid Power icon

### Mode

lite

### Context / Reason

I have ATS for switch automatically V2H and Grid Power.
Please add V2H (vehicle to home) icon and sensors (Hz,V,A) near Grid Power icon.
I see the V2H values ​​via a smart energy sensor

slipx06, 2024-10-19:
HI. Have you tried to use the AUX options to show this? 

[image]

terzo33, 2024-10-19:
> HI. Have you tried to use the AUX options to show this? 
> 
> [image]
> 

Hi @slipx06 , aux option is for load (inverter Output), V2L or V2H is like a grid charge(energy to inverter)

Thanks

slipx06, 2024-10-19:
You can use the AUX port to send energy to the inverter. Some people connect a generator or another micro-inverter. Or you can also use the non-essential circuit. 

This is how some other users have configured it

[image]

https://community.home-assistant.io/t/inverter-power-flow-card/562933/438

terzo33, 2024-10-19:
Thanks a lot