slipx06/sunsynk-power-flow-card #530: On-Grid / Grid tied inverters

Saentist, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

There is a bug when use on-grid inverter.
Home consume x2 energy in graph instead of reality

[image]

```yaml
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
solar:
  show_daily: true
  mppts: 2
  auto_scale: true
  dynamic_colour: true
  max_power: 4200
load:
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: false
  auto_scale: true
entities:
  priority_load_243: sensor.esp8266_limit
  load_frequency_192: sensor.8266_growatt_frequency
  inverter_current_164: sensor.8266_growatt_current_phase_a
  inverter_power_175: sensor.8266_growatt_pv_active_power
  inverter_status_59: sensor.8266_growatt_status_code
  day_battery_discharge_71: sensor.sunsynk_day_battery_discharge
  grid_power_169: sensor.8266_growatt_grid_active_power
  day_grid_import_76: sensor.clamp_electric_meter
  essential_power: none
  nonessential_power: none
  day_pv_energy_108: sensor.esp8266_growatt_today_s_generation
  pv1_power_186: sensor.8266_growatt_pv1_active_power
  pv2_power_187: sensor.8266_growatt_pv2_active_power
  pv1_voltage_109: sensor.8266_growatt_pv1_voltage
  pv1_current_110: sensor.8266_growatt_pv1_current
  pv2_voltage_111: sensor.8266_growatt_pv2_voltage
  pv2_current_112: sensor.8266_growatt_pv2_current
  total_pv_generation: sensor.8266_growatt_total_energy_production
  pv_total: sensor.8266_growatt_power_phase_a
  grid_voltage: sensor.8266_growatt_voltage_phase_a
  energy_cost_buy: sensor.toko_price
  dc_transformer_temp_90: sensor.8266_growatt_inverter_module_temp
  day_load_energy_84: sensor.clamp_meter_daily_consumption
  load_power_L1: sensor.clamp_meter_power
  grid_ct_power_172: sensor.clamp_electric_meter_power
  solar_sell_247: sensor.toko_price
  environment_temp: sensor.zb_temp_humidity_temperature
large_font: false
panel_mode: false
show_battery: false
show_grid: true
dynamic_line_width: true
inverter:
  auto_scale: true
  modern: false
  autarky: power
  model: growatt
```

### Expected behaviour

correcting values

### Possible Solutions

Option to select on-grid inverter

### Mode

Both

### Context / Reason

Option to select on-grid inverter
Entity for grid input
Entity for inverter output

Ignore any hybrid inverter, battery calculations.

slipx06, 2024-10-21:
Hi. Do you have a sensor that measures your home consumption? You should use it for the `essential_power: `
The card will use the formula `essential = inverter_power_175 + grid_power_169 - aux_power_166` if you do not provide this sensor. In your case it is `essential = 450 + 499 = 949W`

Saentist, 2024-10-21:
I have main input and inverter output
System is limited to not export electricity to network

slipx06, 2024-10-21:
You should use your inverter output as your essential_power. Have you tried to set `essential_power: sensor.clamp_meter_power`

Saentist, 2024-10-22:
I set it and will test when sun rises

p.s.
In `Grid entities`, there is no grid voltage.
Under and over some limits all inverters cut the power.

Saentist, 2024-10-22:
Now house consume just imported energy without solar one
need to be `imported + solar`

[image]

guess need to ads something more somewhere 

```yaml
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 0
  shutdown_soc: 20
  show_daily: true
solar:
  show_daily: true
  mppts: 2
  auto_scale: true
  dynamic_colour: true
  max_power: 4200
load:
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: false
  auto_scale: true
entities:
  priority_load_243: sensor.esp8266_limit
  load_frequency_192: sensor.8266_growatt_frequency
  inverter_current_164: sensor.8266_growatt_current_phase_a
  inverter_power_175: sensor.8266_growatt_pv_active_power
  inverter_status_59: sensor.8266_growatt_status_code
  day_battery_discharge_71: sensor.sunsynk_day_battery_discharge
  grid_power_169: sensor.8266_growatt_grid_active_power
  day_grid_import_76: sensor.clamp_electric_meter
  essential_power: sensor.clamp_meter_power
  nonessential_power: none
  day_pv_energy_108: sensor.esp8266_growatt_today_s_generation
  pv1_power_186: sensor.8266_growatt_pv1_active_power
  pv2_power_187: sensor.8266_growatt_pv2_active_power
  pv1_voltage_109: sensor.8266_growatt_pv1_voltage
  pv1_current_110: sensor.8266_growatt_pv1_current
  pv2_voltage_111: sensor.8266_growatt_pv2_voltage
  pv2_current_112: sensor.8266_growatt_pv2_current
  total_pv_generation: sensor.8266_growatt_total_energy_production
  pv_total: sensor.8266_growatt_power_phase_a
  grid_voltage: sensor.8266_growatt_voltage_phase_a
  energy_cost_buy: sensor.toko_price
  dc_transformer_temp_90: sensor.8266_growatt_inverter_module_temp
  day_load_energy_84: sensor.clamp_meter_daily_consumption
  load_power_L1: sensor.clamp_meter_power
  grid_ct_power_172: sensor.clamp_electric_meter_power
  solar_sell_247: sensor.toko_price
  environment_temp: sensor.zb_temp_humidity_temperature
large_font: false
panel_mode: false
show_battery: false
show_grid: true
dynamic_line_width: true
inverter:
  auto_scale: true
  modern: false
  autarky: power
  model: growatt
card_width: "7"
```

slipx06, 2024-10-22:
You can create a template sensor that adds the pv and grid.

```
template:
  - sensor:
      - name: "Essential Power"
        unique_id: essential_power
        device_class: "power"
        unit_of_measurement: "W" 
        state_class: measurement
        state: >
          {% set pv = states('sensor.8266_growatt_power_phase_a') | float %}
          {% set load = states('sensor.clamp_meter_power') | float %}
          {{ (pv + load) |float(0)|round(0) }}
```

Saentist, 2024-10-22:
> You can create a template sensor that adds the pv and grid.
>
> ```
> template:
>   - sensor:
>       - name: "Essential Power"
>         unique_id: essential_power
>         device_class: "power" [...]

and result is 
[image]
https://www.home-assistant.io/faq/unique_id

More simple is to have integrated logic for on-grid inverters
[image]

Saentist, 2024-10-22:
> Well that is not correct. Perhaps you should attempt some troubleshooting. The better option is to provide an essential sensor that measures your inverter output.

I have  entity `sensor.8266_growatt_power_phase_a` for inverter output and it show correctly 
also have `sensor.clamp_electric_meter_power` with show what come in house

[image]

I guess problem is in logic of integration to use hybrid inverter situation, where grid come to inverter and then to house.

slipx06, 2024-10-22:
The template sensor is meant to add 508W + 28W = 536W
If you use this for the essential sensor it should be fine. Have you checked under developer tools that your new sensor is working?
It should not output 0W

Maybe try to create a helper in HA to combine the state of two sensors. 

Sensor A - sensor.8266_growatt_power_phase_a
Sensor B - sensor.clamp_electric_meter_power