slipx06/sunsynk-power-flow-card #533: The version for this plugin can not be used with HACS.

RenierBoshoff, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In HACS, I am trying to download the sunsynk-power-flow-card
 But I get this message:

sunsynk-power-flow-card
Commit will be downloaded

When downloaded, this will be located in '/config/www/community/sunsynk-power-flow-card'
**The version for this plugin can not be used with HACS.**

### Steps To Reproduce

1 . Home Assistant Community Store
2 . Search for sunsynk
3 . Select sunsynk-power-flow-card
4 . Click Download
5 . Click download on Pop up: Commit will be downloaded

When downloaded, this will be located in '/config/www/community/sunsynk-power-flow-card'

6. Gets error: The version for this plugin can not be used with HACS.
[image]

### Expected behaviour

Successful download.

### Card Version

Raspberry Pi 3  rpi3-64

### Home Assistant Version

Home Assistant  Core 2024.10.3 Supervisor 2024.10.3 Operating System 13.2 Frontend 20241002.3

slipx06, 2024-10-26:
Hi. Are you running the latest version of HACS 2.0.1 ?

RenierBoshoff, 2024-10-26:
> Hi. Are you running the latest version of HACS 2.0.1 ?

Hi!

Yes I am running the latest version.
After a reboot and some fiddling with DNS server settings I got it loaded.
[image]

Now I am at the next step, trying to get this Wave USB to RS485 Industrial Converter [https://www.robotics.org.za/W17286]
to collect the direct RS485 inverter data and display it on your Flow card...

I feel like I have no idea what I am doing even after watching a lot of videos, and reading up on the settings.

Do I need to paste this somewhere: [https://github.com/slipx06/Sunsynk-Home-Assistant-Dash/blob/main/Dashboard.yaml](url)

RenierBoshoff, 2024-10-26:
By the way, this is the physical connections:

Inverter RS485 Port > RS485 to USB converter (the WaveShare one) > USB port of Raspberry.