slipx06/sunsynk-power-flow-card #534: 16:9 layout for the card

LucianTothezan, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

the card layout is a square so using it as a single card in Panel layout on a tablet would not display the card on the whole area of the 16:9 display.

### Expected behaviour

I would like to use this card as Panel on a wall mount tablet so the card should cover the whole display area. Therefore a 16:9 and/or 9:16 layout would be great.

### Mode

Both

### Context / Reason

wallmount 16:9 tablet with this card as Panel display.

slipx06, 2024-10-28:
The card is rendered as an SVG element with a viewbox that is set to preserve aspect ratio . At this stage there is no quick fix. You can try adjust the card_height: , card_width: and panel: attributes. You can also experiment with different dashboard layouts i.e. grid layout etc

LucianTothezan, 2024-10-28:
Thank You for the reply. I really really appreciate this card. I have
already tried what you suggested but unfortunately the result remains the
same: if panel mode I can modify the height but it will auto adjust also
the width. In grid mode also, it will modify the width but also
automatically will keep the aspect ratio. It would really need an 16:9 SVG
that way if we modify the width or the height, it would preserve the native
wide format of the SVG. Unfortunately I do not have the proper knowledge to
create such an element by myself....

Best regards,

Lucian Tothezan

PR #556 (2024-11-17): New wide screen view

## Description

Adds and wide screen toggle to display card in 16:9 format
## Related issues

Closes #551
Closes #534 
## Motivation and Context

A wide screen view allows the display of two batteries as well as multiple inverters

## Type of change
New feature (non-breaking change which adds functionality)