slipx06/sunsynk-power-flow-card #535: Support for translation into Ukrainian

lightzhuk, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The language of the system is Ukrainian, the text of the map is in English

### Expected behaviour

Automatic translation into the selected system language

### Mode

Both

### Context / Reason

Maybe I can help with the translation? but I'm not good at code

slipx06, 2024-10-26:
Hi. Can you assist with translating the text below. You can see examples [here src/localize/languages](https://github.com/slipx06/sunsynk-power-flow-card/tree/8640b9662e2576688952e01a25ffad73ff377d79/src/localize/languages)

```
{
  "common": {
    "description": "A Home Assistant card to emulate the Sunsynk power flow that's shown on the Inverter screen",
    "version": "Version",
    "runtime_to": "RUNTIME TO",
    "battery_floating": "BATTERY FLOATING",
    "charge": "CHARGE",
    "to": "TO",
    "daily_load": "DAILY LOAD",
    "daily_aux": "DAILY AUX",
    "daily_grid_buy": "DAILY GRID BUY",
    "daily_grid_sell": "DAILY GRID SELL",
    "daily_solar": "DAILY SOLAR",
    "daily_solar_left": "DAILY SOLAR / LEFT TODAY",
    "total_solar_generation": "DAILY SOLAR / TOTAL SOLAR",
    "daily_charge": "DAILY CHARGE",
    "daily_discharge": "DAILY DISCHARGE",
    "ratio": "Ratio",
    "timer_off": "Timer off",
    "timer_on": "Timer on",
    "priority_batt": "Priority Batt",
    "priority_load": "Priority Load",
    "nonessential_name": "Non Essential",
    "aux_name": "Auxiliary",
    "pv1_name": "PV1",
    "pv2_name": "PV2",
    "pv3_name": "PV3",
    "pv4_name": "PV4",
    "essential": "Essential",
    "autarky": "Autarky",
    "discharging": "Discharging",
    "charging": "Charging",
    "waiting": "Waiting",
    "standby": "Standby",
    "no_battery": "No Battery",
    "selftest": "Selftest",
    "normal": "Normal",
    "alarm": "Alarm",
    "fault": "Fault",
    "days": "days",
    "hrs": "hrs",
    "min": "min",
    "flash": "Flash",
    "check": "Check",
    "idle": "Idle",
    "exporting": "Export",
    "importing": "Import",
    "ongrid": "On-Grid",
    "offgrid": "Off-Grid",
    "normalstop": " Normal Stop",
    "shutdown": "Shutdown",
    "offline": "Offline",
    "running": "Running",
    "sleepmode": "Sleep Mode",
    "grid_name": "Grid",
    "limit": "Limit",
    "off": "Off",
    "lowpower": "Low Power",
    "bulk": "Bulk",
    "absorption": "Absorption",
    "float": "Float",
    "storage": "Storage",
    "equalize": "Equalize",
    "passthru": "Passthru",
    "inverting": "Inverting",
    "powerassist": "Power Assist",
    "powersupply": "Power Supply",
[... 117 lines cut ...]
    "label_daily_grid_sell": "Daily Grid Sell Label",
    "invert_flow": "Invert Flow",
    "label_daily_load": "Daily Load Label",
    "cat_title": {
      "title": "Title Options",
      "general": "General Options",
      "inverter": "Inverter Options",
      "solar": "Solar Options",
      "battery": "Battery Options",
      "grid": "Grid Options",
      "load": "Load Options",
      "entities": "Entities",
      "aux": "Aux Options",
      "energy": "Energy Entities",
      "bat": "Battery Entities",
      "sol": "Solar Entities",
      "inv": "Inverter Entities",
      "ld": "Load Entities",
      "gri": "Grid Entities",
      "sensor": "Optional Sensors"
    }
  }
}

```

lightzhuk, 2024-10-26:
I will be happy to do so in the near future

slipx06, 2024-10-30:
Thank you. Some of the text is a bit long. Are there perhaps shorter words or abbreviations we can use?

[image]

  ```
    "priority_batt": "Пріоритет Батареї",
    "priority_load": "Пріорітет Навантаження",

```

slipx06, 2024-10-30:
I will use this

```
"priority_batt": "Пріор. Бат."
"priority_load": "Пріор. Наван."
```

closed 2024-10-30 by slipx06

amgstone, 2024-10-30:
[image]

slipx06, 2024-10-30:
Yes need some help with a alternative shorter translation for these two

"autarky": "Самодостатність",
"ratio": "Співвідношення",

amgstone, 2024-10-30:
"ratio": "Пропорція",

amgstone, 2024-10-30:
daily gred buy- З мережі за день
daily load- Спожито за день

amgstone, 2024-10-30:
залишок сьогодні- Прогноз на сьогодні

slipx06, 2024-10-30:
Thanks. Daily Load and Daily Grid Buy are already included in the translation. You just need to reset your front end cache or force a reload as both of those can be customized and for some reason the first time you switch languages do not update

commit 706f6f14 (2024-10-30): feat: add support for Sungrow inverters. closes #547
locale: add Ukraine translation. closes #535