slipx06/sunsynk-power-flow-card #536: I do not consume electricity from the network, but it shows that there is consumption

amgstone, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I do not consume electricity from the network, but it shows that there is consumption
[image]

### Expected behaviour

I do not consume electricity from the network, but it shows that there is consumption

### Card Version

new

### Home Assistant Version

new

slipx06, 2024-10-27:
Hi.
According to your sensor readings you are consuming more power than you are producing. 
721W of Solar power, but consuming 755W (414W +341W) from battery and load. Are your sensors accurate? Are you providing your own sensor for essential_power?

amgstone, 2024-10-27:
show_daily_sell: false
entities:
  inverter_voltage_154: none
  load_frequency_192: none
  inverter_current_164: none
  inverter_power_175: none
  grid_power_169: sensor.smg_output_power
  battery_soc_184: sensor.jk_bms_state_of_charge
  battery_power_190: sensor.jk_bms_power
  grid_ct_power_172: sensor.smg_mains_power
  grid_connected_status_194: binary_sensor.energy_in_sensor
  no_grid_colour: green
  inverter_status_59: sensor.smg_ii_operation_mode
  pv1_power_186: sensor.smg_ii_pv_average_power
  pv2_power_187: sensor.smg_ii_2_pv_average_power
  pv1_voltage_109: none
  pv1_current_110: none
  pv2_voltage_111: none
  pv2_current_112: none
  day_pv_energy_108: sensor.soniachni_paneli
  day_battery_charge_70: sensor.bms0_charging_energy_today
  day_battery_discharge_71: sensor.bms0_discharging_energy_today
  day_load_energy_84: sensor.solar_plus_energy_minus_battery_sensor
  day_grid_import_76: sensor.spozhito_elektroenergiyi_za_sogodni
  essential_load1: sensor.oncharger_power
  essential_load2: sensor.ten_boiler_power
  essential_load3: sensor.fermi_power
  essential_load4: sensor.recuperator_power
  remaining_solar: sensor.prognoz_na_sogodni
large_font: false

slipx06, 2024-10-28:
You should provide an `essential_power:` sensor that measure your house load

amgstone, 2024-10-29:
in entities: add essential_power: sensor.smg_output_power
[image]

slipx06, 2024-10-30:
What inverter are you using and which integration into HA?