slipx06/sunsynk-power-flow-card #539: sunsynk-power-flow-card not working in home assistant mobile app

basti122303, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When home assistant is > 2024.7.4 sunsynk-power-flow-card stop working in mobile app
Web browser like Chromium is working.

- App Version: 2024.10.3-full
- HA Version: 2024.10.4

[image]

### Steps To Reproduce

1. Install home assistant > 2024.7.4
2. Install sunsynk-power-flow-card
3. Try to open sunsynk-power-flow-card in mobile app

### Expected behaviour

Work like in web browser

### Card Version

5.3.1

### Home Assistant Version

2024.10.4

slipx06, 2024-10-28:
Hi This is related to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v5.0.0

closed 2024-10-28 by slipx06

basti122303, 2024-10-28:
- HA is up to date
- sunsynk-power-flow-card is up to date
- mobile app is up to date

So how should I fix this issue?

slipx06, 2024-10-28:
I'm not sure. You are running the latest version of the companion app. Have you tried to clear the app cache?

basti122303, 2024-10-28:
I have clear the cache / user data also try to reinstall the app nothing solved the issue.
On the browser (Firefox / Chromium) all is fine.
In the app also all is fine except sunsynk-power-flow-card

slipx06, 2024-10-28:
I don't know what else to try. What mobile device and android version are you using? Perhaps other users can help

basti122303, 2024-10-28:
samsung with android 10, OK Thx, I will try some other versions

basti122303, 2024-11-01:
I did some more testing, all mobile apps are 2024.10.3-full, HA and Sunsynk Power Flow card are up to date
- Android 14 works
- Android 7 works
- Android 10 (LineageOS) doesn't work in mobile app or browser. I can see the screenshot as above