slipx06/sunsynk-power-flow-card #545: add options to essential power

peter-j-cornell, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

positive essential power via Solar Assistant becomes negative when shown on power flow card .  Also flow dots show power being imported.

### Expected behaviour

Either: essential power flow is automatically shown as consumption
Or: there is an option to invert the power flow as with other loads

### Mode

Both

### Context / Reason

Use: Power flow card being used with SolarSynk inverters via MQTT and Solar Assistant.
Reason: to tidy up the display so that power flows in the correct direction

slipx06, 2024-10-28:
Hey. Did you try the toggle below

[image]

peter-j-cornell, 2024-10-28:
> Hey. Did you try the toggle below
>
> image.png (view on web) <https://github.com/user-attachments/assets/e6269218-36f0-42cd-a518-7d4530234e31>

Thanks for quick reply

The toggle you have shown me toggles the non-essential load power as well as the essential load power.  In my set-up the non-essential load power and the essential load power have opposite signs when consuming power.  I realise this is unusual.

Peter

closed 2024-10-28 by slipx06