slipx06/sunsynk-power-flow-card #550: Suggestion

Hans-chr2, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

For a lack of finding a suggestion box i have created this feature request. I use the card alot for my Growatt system and i absolutely love it. 

### Expected behaviour

none

### Possible Solutions

I recently integrated my district heating meter into my Home Assistant. To visualize the consumption, I would really like a "district heating energy flow card." I imagine that the card could show something similar to the draft I've made. There are many elements from the Sunsynk card that I think could also work well on a new card. Unfortunately, I don’t have the skills to make such a card myself.

[image]

### Mode

lite

### Context / Reason

I guess there is alot of people out there that needs a card like this. I have searched google high and dry for a card that look in any way similar but with no luck. Then i though about power flow card, and there is alot of elements that can be used. 

I if this request falls way out of line, fell free to delete it. :-)

slipx06, 2024-11-18:
Hi 
Have a look at the picture elements card. It could work for this use case
https://www.home-assistant.io/dashboards/picture-elements/

Hans-chr2, 2024-11-25:
Hi Slipx06

I already have looked at it, but it like the animation and general layout from your card alot better. :-) 
So i though 90% of the work was all ready done in your. The only addition would we a second line from Grid (boiler) to House. 

You are free to delete the thread if you like. 👍

closed 2024-12-08 by slipx06