slipx06/sunsynk-power-flow-card #552: Is it possible to display 3 inverters in parallel ?

jean-luc1203, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

If I've looked hard enough, I can't display several inverters on the board as well as 6 mppt.
In my case, 3 WKS inverters.

### Expected behaviour

It would be nice to be able to **display 3 inverters** (or a box with the 3 values) with their 2 MPPT inputs. 
At the output of these 3 machines, display the total power to the house load. 
I obviously have the power supplied by each of the inverters.
Is this conceivable?
That said, **many thanks** for all the work you've done for us.

### Mode

lite

### Context / Reason

[image]

slipx06, 2024-11-12:
Not at the moment. You could display each inverter on there own card or create template sensors to sum the outputs

jean-luc1203, 2024-11-12:
thank you for your reply. 
I do indeed have all the entities already present, it's simply a visual desire to be able to see 3 inverters with their respective production at a glance.
Maybe I'll use 3 PV inputs to simulate 3 inverters and the sum of the 3 will go to the board's central inverter...
I'll have to see if it's readable like that.