slipx06/sunsynk-power-flow-card #554: Remaining battery energy

flopp999, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]
It says remaining is 3.9 kWh but actually I only have 0.9 kWh left because my battery will stop discharge at 10%.
I have set my SoC end to 10%. I have 30000 Wh.

### Expected behaviour

Showing 0 kWh remain when SoC is at 10%

### Possible Solutions

I searched through the code but I couldn't find where the math is done. But remove SoC End before calculate the remaining kWh

### Mode

lite

### Context / Reason

So users don't think there is more kWh than there are

flopp999, 2024-11-15:
Found it in the code, I will try to make the math as I asking for
https://github.com/slipx06/sunsynk-power-flow-card/blob/706f6f14dbdee815c36edee5457148f7c081c33e/src/cards/full-card.ts#L343

slipx06, 2024-11-18:
Hi
At the moment the remaining energy is calculated based on total battery capacity and not shutdown SOC.

flopp999, 2024-12-07:
OK, what do you think of changing that?
I fix until that will be implemented is to change the shutdown SoC to 0% and the amount of energy

slipx06, 2024-12-08:
Le me think about adding both options

commit a2304183 (2024-12-18): feat: adds `off_colour:` and `remaining_energy_to_shutdown:` closes #577 closes #554
closed 2024-12-18 by slipx06