slipx06/sunsynk-power-flow-card #557: Upgrade to version 6 keeps failing

Supermanenvysme, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Have been trying to upgrade to version 6 (from version 5) and when replacing the js, it fails on the dashboard. Roll back to v5 and it restores. tried a few times since yesterday but unsuccessful.

### Expected behaviour

to initialise once the JS file is replaced and refreshed

### Card Version

6

### Home Assistant Version

2024.11.2

slipx06, 2024-11-18:
Hi
Try upgrade using HACS? 
Any browser logs?

Supermanenvysme, 2024-11-19:
This is the error I receive as soon as the js file is replaced:

Custom element not found: sunsynk-power-flow-card

slipx06, 2024-11-19:
Probably an issue with you browser caching old content. I cant really help without logs. You should clear your browser cache. Also the preferred way to install this is using HACS.

Supermanenvysme, 2024-11-20:
Refreshed the cache a few time.

Error from HA:

Failed to execute 'define' on 'CustomElementRegistry': the name "sunsynk-power-flow-card-dev" has already been used with this registry window.CustomElementRegistry.define (src/scoped-custom-element-registry.js:44:14) /local/sunsynk-power-flow-card.js:30:100 e (/local/sunsynk-power-flow-card.js:1:250) /local/sunsynk-power-flow-card.js:46:684457

Probably going to stay on ver5 if this cant be resolved.

slipx06, 2024-11-20:
Looks like you are trying to load a development version of the card?  You should remove all instances of sunsynk-power-flow-card.js. Then do a clean install.

You should see this in your browser logs

[image]

NOT

[image]