slipx06/sunsynk-power-flow-card #558: Custom font size or automatic font size

LucianTothezan, 2024-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Font size makes it th
at for main loads the text is overlaping.

### Expected behaviour

Automatic font adjustement or possibility to manually set font size

### Mode

full

### Context / Reason

[image]

slipx06, 2024-11-19:
Hi, The work around here is to use abbreviated load names or move them around. i.e. swap Bucataria with Baie. 
Making the font smaller wont really help as it would become too small. I'll make a note, but this is not planned now.

LucianTothezan, 2024-11-19:
thank You for the feedback. i will keep that in mind :)