slipx06/sunsynk-power-flow-card #560: How to get this Live Home Usage box to 'click through' to a sensor?

richsimpson, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Most or all of the values click through to the relevant sensor for history. The current usage of the house doesnt. 
I don;t see that item as a value in the setup. 

[image]

I think I need to add the sensor somewhere ( sensor.solax_house_load ) but currently the value is being calculated based on what is left from a calculation?

Can you help with what I add and where please?

### Expected behaviour

Would like to see the sensor data if click on the box 
sensor is sensor.solax_house_load 

### Card Version

6.0.0

### Home Assistant Version

2024.11.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
show_grid: true
show_battery: true
battery:
  energy: 17600
  shutdown_soc: 10
  show_daily: true
  invert_power: true
  max_power: 3750
  auto_scale: true
  show_absolute: false
  hide_soc: true
  dynamic_colour: true
  linear_gradient: false
  show_remaining_energy: true
  animate: false
solar:
  show_daily: true
  mppts: 1
  max_power: 7000
  display_mode: 2
  animation_speed: 9
  dynamic_colour: true
  auto_scale: true
load:
  show_daily: true
  max_power: 15000
  show_aux: false
  show_nonessential: true
  additional_loads: 2
  load1_name: Water
  load2_name: EV
  load1_icon: mdi:thermometer-water
  load2_icon: mdi:ev-station
  animation_speed: 9
  dynamic_colour: true
  auto_scale: true
  dynamic_icon: true
  invert_load: false
  load3_name: UFH
  load3_icon: ""
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  animation_speed: 9
  auto_scale: false
  export_colour:
    - 187
    - 91
    - 200
  dynamic_colour: true
  no_grid_colour:
    - 189
    - 188
    - 188
entities:
  inverter_voltage_154: sensor.solax_inverter_voltage
  inverter_current_164: sensor.solax_inverter_current
  inverter_power_175: sensor.solax_inverter_power
  inverter_status_59: sensor.solax_run_mode
  radiator_temp_91: sensor.solax_inverter_temperature
  day_battery_charge_70: sensor.solax_battery_input_energy_today
  day_battery_discharge_71: sensor.solax_battery_output_energy_today
  battery_voltage_183: sensor.solax_battery_voltage_charge
  battery_soc_184: sensor.solax_battery_capacity
  battery_power_190: sensor.solax_battery_power_charge
  battery_current_191: sensor.solax_battery_current_charge
  battery_temp_182: sensor.solax_battery_temperature
  day_grid_import_76: sensor.solax_today_s_import_energy
  day_grid_export_77: sensor.solax_today_s_export_energy
  grid_power_169: sensor.solax_grid_export_import_sum2
  grid_ct_power_172: sensor.solax_grid_export_import_sum2
  day_load_energy_84: sensor.powerflow_today_house_load
  day_pv_energy_108: sensor.solax_today_s_solar_energy
  pv1_power_186: sensor.solax_pv_power_1
  pv1_voltage_109: sensor.solax_pv_voltage_1
  pv1_current_110: sensor.solax_pv_current_1
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  essential_load1_extra: sensor.immersion_water_ct3_today
  essential_load2: sensor.ev_fast_charge_ct1
  essential_load2_extra: sensor.ev_fast_charge_ct1_day
  essential_load1: sensor.immersion_current_power_ct3
large_font: false
panel_mode: false
inverter:
  auto_scale: false
  modern: false
  model: solax
  autarky: "no"
```

wacalitz, 2024-11-24:
Check the Entities section > Load Entities > Essential_power 

[image]

For reference this is what mine is set to on the sunsynk setup.
  essential_power: sensor.sunsynk_load_power_ess
  essential_load1: sensor.sonoff_geyser_power
  essential_load2: sensor.sonoff_stove_power

Looks like you haven't set the sensor for it as mine does give the history pop up you are looking for.

richsimpson, 2024-11-24:
> Check the Entities section > Load Entities > Essential_power [...]

Spot on, that's solved it.

closed 2024-11-24 by richsimpson