slipx06/sunsynk-power-flow-card #563: Orange Dot left top corner in Compact and Lite view

ragman11, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In the Comapct and Lite View there is an orange dot in the left top corner. 

### Expected behaviour

how can I remove this dot
[image]

### Card Version

latest as of 25. november 2024

### Home Assistant Version

latest as of 25.november 2024

slipx06, 2024-11-25:
Hi. Please share you config

ragman11, 2024-11-25:
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
battery:
  energy: 2880
  show_daily: true
  show_absolute: true
  auto_scale: true
  show_remaining_energy: true
  max_power: 600
  charge_colour: "#f06292"
  colour: "#4db6ac"
  shutdown_soc: 10
solar:
  show_daily: true
  pv1_name: GHV
  pv2_name: GHH
  auto_scale: true
  dynamic_colour: true
  max_power: 1500
  pv1_max_power: 400
  pv2_max_power: 300
  mppts: 4
  pv3_name: SF
  pv4_name: FLEX
  pv3_max_power: 800
  pv4_max_power: 300
load:
  show_daily: true
  essential_name: MoleNord
  dynamic_colour: true
  load1_name: E-Charge
  load1_icon: mdi:ev-station
  additional_loads: 4
  load2_name: HeatSF
  load2_icon: mdi:heat-wave
  auto_scale: true
  dynamic_icon: true
  max_power: 2000
  load3_name: KASPA
  load3_icon: mdi:bitcoin
  load4_icon: mdi:folder-home-outline
  load4_name: HO/TV
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  show_essential: true
  nonessential_name: Export
  auto_scale: true
  max_power: 2000
  export_colour: "#DF4C1E"
entities:
  inverter_power_175: sensor.solar_power_shelly_power
  day_battery_charge_70: sensor.sfbatterychargedaily
  day_battery_discharge_71: sensor.sfoutputbattdaily
  battery_soc_184: sensor.electriclevel
  battery_power_190: sensor.batteryposneg
  day_grid_import_76: sensor.energy_import_daily
  day_grid_export_77: sensor.energy_export_daily
  grid_ct_power_172: sensor.import_posneg
  grid_power_169: sensor.import_posneg
  day_load_energy_84: sensor.energy_consumption_daily
  day_pv_energy_108: sensor.solarshellydaily
  pv1_power_186: sensor.ghv_power
  pv3_power_188: sensor.solarinputpower
  pv2_power_187: sensor.ghh_power
  nonessential_power: sensor.power_export
  essential_power: sensor.power_consumption
  battery_current_191: none
  essential_load1: sensor.overpower_plug_power
  essential_load2: sensor.brunnen_power
  environment_temp: sensor.netatmo_in_wohnzimmer_garten_temperature
  battery_temp_182: sensor.batterytempall
  energy_cost_buy: sensor.epex_spot_data_net_price
  essential_load3: sensor.shelly_luckyminer_power
  essential_load4: sensor.homeofficetvpower
  pv1_voltage_109: sensor.ghv_ch1_voltage
  pv2_voltage_111: sensor.ghh_ch1_voltage
  pv2_current_112: sensor.ghh_ch1_current
  pv1_current_110: sensor.ghv_ch1_current
  pv4_power_189: sensor.flex_power
  pv4_voltage_115: sensor.flex_ch1_voltage
  pv4_current_116: sensor.flex_ch1_current
inverter:
  autarky: energy
  auto_scale: true
large_font: true
dynamic_line_width: true
max_line_width: 4
wide: false

ragman11, 2024-11-25:
the dot is only there when Solar power is available.

slipx06, 2024-11-25:
If you set `show_solar: false` does it go away. What browser are you using?

ragman11, 2024-11-25:
I'm using Brave and Chrome browser, with show_solar: false the dot is gone

slipx06, 2024-11-25:
I cant replicate this. Can you try removing your PV entities one at a time and tell me if it goes away at all

```
pv1_voltage_109: sensor.ghv_ch1_voltage
pv1_current_110: sensor.ghv_ch1_current
pv2_voltage_111: sensor.ghh_ch1_voltage
pv2_current_112: sensor.ghh_ch1_current
pv4_power_189: sensor.flex_power
pv4_voltage_115: sensor.flex_ch1_voltage
pv4_current_116: sensor.flex_ch1_current
```

slipx06, 2024-11-25:
Can you confirm this is resolve in 6.1.2

closed 2024-11-25 by slipx06

ragman11, 2024-11-25:
Tanks for the Fast fix, witzig 6.1.2 the dot is gone.

slipx06, 2024-11-25:
Thanks for catching that

commit cc0c80d7 (2024-11-25): fix: pv3 animation closes #563