slipx06/sunsynk-power-flow-card #565: Home Production Anomaly

MaurizioBarbera, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

You use solar production and no inverte output. the HUAWEI SUN2000 it has two sensors for the inverter one input power to the inverter and another which is the active power (output), it is possible to use in the calculation: inverte_power_175 and not pv1_power_186...

doc: Home Production = Solar + Battery (discharge) + Aux (in)
change on: Home Production = inverter_power_175 + Battery (discharge) + Aux (in)
thanks and confident in your exhaustive answer the question

### Expected behaviour

Ratio and autarky more precise

### Card Version

6.1.3

### Home Assistant Version

2024.11.0

slipx06, 2024-12-06:
Hi. That logic is only used to display the additional Autarky and Ratio stats. Removing Solar or creating a custom formula does not really make sense for the majority of users. You should rather hide that information if it is inaccurate for your setup by setting `autarky: no`

MaurizioBarbera, 2024-12-08:
hi, okay thanks

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

closed 2024-12-08 by slipx06