slipx06/sunsynk-power-flow-card #566: Grid_CT_Power is zero for SolArk (Deye)

chicagoandy, 2024-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Overall problem is Grid power is showing zero while charging/discharging
[image]

The card is using Grid_CT_Power (172) as it's listed as required field for Inverter Entities. but this appears wrong for my Sol-Ark.

[image]

Grid_power (169) seems like a better choice?

[image]

Is the documentation out-of-date?

I have specified   grid_power_169: sensor.sunsynk_grid_power, but that does not seem to be helping.

### Expected behaviour

Grid power should be visible in the card.

### Card Version

6.1.3

### Home Assistant Version

2024.11.3

### Configuration

```shell
- platform: modbus_controller            # 172 Grid CT Power
    modbus_controller_id: inverter
    name: "${friendly_name} Grid CT Power"
    id: inverter_esphome_grid_ct_power
    register_type: holding
    address: 172
    unit_of_measurement: "W"
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    value_type: S_WORD

  - platform: modbus_controller            # 169 Grid Power
    modbus_controller_id: inverter
    name: "${friendly_name} Grid Power"
    id: inverter_esphome_grid_power_169
    register_type: holding
    address: 169
    unit_of_measurement: "W"
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    value_type: S_WORD

type: custom:sunsynk-power-flow-card
cardstyle: lite
large_font: true
show_solar: true
panel_mode: true
card_height: 415px
inverter:
  modern: true
  colour:
    - 218
    - 215
    - 215
  autarky: power
  model: deye
  auto_scale: true
solar:
  show_daily: true
  colour: "#F4C430"
  animation_speed: 9
  max_power: 10000
  pv1_name: West
  pv2_name: South
  display_mode: 2
  auto_scale: true
  pv2_max_power: 3660
  pv1_max_power: 5850
  efficiency: 3
  mppts: 3
  pv3_name: Porch
  dynamic_colour: true
  pv3_max_power: 1000
battery:
  energy: 51200
  shutdown_soc: 5
  show_daily: true
  colour: pink
  animation_speed: 6
  max_power: 13000
  auto_scale: true
  show_absolute: true
  animate: true
  path_threshold: 90
  count: 1
  soc_end_of_charge: 100
  dynamic_colour: true
  linear_gradient: true
  show_remaining_energy: true
load:
  show_aux: false
  show_daily: true
  animation_speed: 8
  max_power: 15000
  load1_name: HVAC
  load1_icon: mdi:air-conditioner
  load3_name: Cothes Dryer
  load3_icon: mdi:tumble-dryer
  auto_scale: true
  path_threshold: 90
  additional_loads: 3
  load2_name: Tesla
  load2_icon: mdi:car
  off_threshold: 3
  dynamic_colour: false
grid:
  show_daily_buy: true
  no_grid_colour: green
  animation_speed: 8
  max_power: 60000
  auto_scale: true
  grid_off_colour:
    - 122
    - 0
    - 0
  grid_name: Georgia Power
  label_daily_grid_buy: Imported Power
  energy_cost_decimals: 2
  import_icon: mdi:transmission-tower-import
  disconnected_icon: mdi:transmission-tower-off
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  day_battery_charge_70: sensor.sunsynk_day_battery_charge
  day_battery_discharge_71: sensor.sunsynk_day_battery_discharge
  day_load_energy_84: sensor.sunsynk_day_load_energy
  day_grid_import_76: sensor.sunsynk_day_grid_import
  day_grid_export_77: sensor.sunsynk_day_grid_export
  day_pv_energy_108: sensor.sunsynk_day_pv_energy
  inverter_voltage_154: sensor.sunsynk_inverter_voltage
  load_frequency_192: sensor.sunsynk_load_frequency
  inverter_current_164: sensor.sunsynk_inverter_current
  inverter_power_175: sensor.sunsynk_inverter_power
  grid_power_169: sensor.sunsynk_grid_power
  pv1_power_186: sensor.sunsynk_pv1_power
  pv2_power_187: sensor.sunsynk_pv2_power
  pv3_power_188: sensor.sunsynk_pv3_power
  pv4_power_189: none
  battery_voltage_183: sensor.sunsynk_battery_voltage
  battery_soc_184: sensor.sunsynk_battery_soc
  battery_power_190: sensor.sunsynk_battery_power
  battery_current_191: sensor.sunsynk_battery_current
  essential_power: sensor.sunsynk_essential_power_1
  essential_load1: sensor.hvac_power_2
  essential_load3: sensor.clothes_dryer_power
  nonessential_power: sensor.sunsynk_nonessential_power_1
  pv1_voltage_109: sensor.sunsynk_pv1_voltage
  pv1_current_110: sensor.sunsynk_pv1_current
  pv2_voltage_111: sensor.sunsynk_pv2_voltage
  pv2_current_112: sensor.sunsynk_pv2_current
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  aux_power_166: sensor.sunsynk_aux_power
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  battery_temp_182: sensor.sunsynk_battery_temperature
  radiator_temp_91: sensor.sunsynk_radiator_temperature
  dc_transformer_temp_90: sensor.sunsynk_dc_transformer_temperature
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_prog1_capacity
  prog1_charge: switch.sunsynk_prog1_grid_charge
  prog2_time: sensor.sunsynk_time_slot_2
  prog2_capacity: number.sunsynk_prog2_capacity
  prog2_charge: switch.sunsynk_prog2_grid_charge
  prog3_time: sensor.sunsynk_time_slot_3
  prog3_capacity: number.sunsynk_prog3_capacity
  prog3_charge: switch.sunsynk_prog3_grid_charge
  prog4_time: sensor.sunsynk_time_slot_4
  prog4_capacity: number.sunsynk_prog4_capacity
  prog4_charge: switch.sunsynk_prog4_grid_charge
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.sunsynk_prog5_capacity
  prog5_charge: switch.sunsynk_prog5_grid_charge
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.sunsynk_prog6_capacity
  prog6_charge: switch.sunsynk_prog6_grid_charge
  environment_temp: sensor.home_realfeel_temperature
  essential_load2: sensor.sensor_tesla_wall_connector_total_power
  pv3_voltage_113: sensor.sunsynk_pv3_voltage
  pv3_current_114: sensor.sunsynk_pv3_current
  energy_cost_buy: sensor.electricity_pricesensor.electricity_price
  pv_total: sensor.sunsynk_solar_power
  battery_soh: sensor.ruixu_battery_monitor_bank_state_of_health
  grid_voltage: sensor.sunsynk_grid_voltage
  grid_ct_power_L2: sensor.sunsynk_load_l2_power
dynamic_line_width: true
show_battery: true
show_grid: true
grid_options:
  columns: 24
  rows: 7
wide: true
view_layout:
  grid-area: flow
```

slipx06, 2024-11-29:
Hi
Do you have a CT coil installed? Those registers are correct for my system but don't get hung up on the names. You should simply provide the correct sensor for grid. Add the following to your list of entities

`grid_ct_power_172: sensor.sunsynk_grid_power`

chicagoandy, 2024-11-29:
I don't have CT sensors installed.   That sensor is always zero as a result.

What sensor should I substitute?   Grid_power (169) ?

slipx06, 2024-11-29:
Yes.  Like I said use grid_ct_power_172: sensor.sunsynk_grid_power

closed 2024-12-17 by slipx06