slipx06/sunsynk-power-flow-card #571: Customizable sensor for Daily Aux and add current,voltage and frequency values to AUX Load

terzo33, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Daily aux is calculated by card and i don't know why is always 0, but i prefer control daily load by my sensor like in grid option, then i can't see current,voltage and frequency values for aux load

### Expected behaviour

I need to calculate daily aux production by my accurate sensor daily. Then i need to see in my card "aux load" current,voltage and frequency values

### Possible Solutions

Add customizable daily aux production and in aux load add current,voltage and frequency values

[image]

[image]

### Mode

lite

### Context / Reason

Check accurate values

slipx06, 2024-12-07:
Hi. Daily Aux is not calculated by the card. To display Daily Aux you need to provide a sensor i.e

`day_aux_energy: sensor.my_daily_aux`

terzo33, 2024-12-07:
> Hi. Daily Aux is not calculated by the card. To display Daily Aux you need to provide a sensor i.e
> 
> `day_aux_energy: sensor.my_daily_aux`

do i have to manually create the sensor with that name or is there a section in the card where i can put my sensor?. And for current,voltage and frequency "AUX" values ?

Thanks

slipx06, 2024-12-07:
Hi. You need to provide your own sensor for that entity. There is no option to display those other values at this stage. 

[image]

terzo33, 2024-12-07:
@slipx06 thanks a lot. Could you improve(current,voltage and frequency values for aux load) in next release?

slipx06, 2024-12-08:
Not at the moment but moving this to an ideas discussion