slipx06/sunsynk-power-flow-card #576: Adding Energy use to all loads

danielTinkiel, 2024-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

This issue might be a product of my own incompetence/stupidity but I fail to see where, in the visual set up interface, I can choose to add the energy used  and status (for example "Connected") by the individual loads. If that feature exists, please forgive me and let me know how to resolve it in my case. I have attached the configuration code editor file below and a copy of my screenshot. My card is set up as "**Compact**".
[image]
[SunSynk1.txt](https://github.com/user-attachments/files/18135592/SunSynk1.txt)

### Expected behaviour

I would like to see in my card the daily energy used by card charger and my hot water diverter in the same way as I see it for the whole load

### Possible Solutions

So far, I only attempted to patch the situation by creating an additional card in he same dashboard displaying it there. Please see the screenshot attached.
[image]

### Mode

lite

### Context / Reason

As your card is great to integrate on one screen both, power and energy, I think it helps to provide a more concise view to the people interested in the detail. Also having the possibility of including a dynamic icon which changes in example colour based on the status (connected or disconnected) provides a much more augmented representation of the current status at a glance

slipx06, 2024-12-15:
Hi. You can display  additional information for each of the essential loads. This can be a sensor that calculates your daily energy.

You can add these under your entities section
```
essential_load1_extra:
essential_load2_extra:
essential_load3_extra:
essential_load4_extra:
essential_load5_extra:
essential_load6_extra: 
```
or through the visual editor

[image]

[image]

danielTinkiel, 2024-12-15:
Hi,
Many thanks for your quick response.
I was looking for that in the visual editor but I cannot find it in my currently used version v6.1.7. Please see attached what I can see in the visual editor. So, I suppose that unless there is something else in the visual editor that I did not see, the option to be used is to include the additional daily energy sensors in the load entities section. Please let me know whether I am missing something. Many thanks again.
Kind regards,
Daniel

Daniel Tinkiel
10 King Edward Park,
Bray, Co. Wicklow, A98EC52
Ireland
+353 86 382 3929

image.png (view on web)<https://github.com/user-attachments/assets/a3911e32-9245-44c4-af7a-3fdcc852a156>

image.png (view on web)<https://github.com/user-attachments/assets/33267ccc-4656-4222-b8ae-e1456f189307>

danielTinkiel, 2024-12-15:
Hi,
Please disregard my last email. I found these fields in the load>entities rather than "load options" which was where I was looking at.
Kind regards,
Daniel

Daniel Tinkiel
10 King Edward Park,
Bray, Co. Wicklow, A98EC52
Ireland
+353 86 382 3929

closed 2024-12-17 by slipx06