slipx06/sunsynk-power-flow-card #577: Hide additional loads when power is below limit (0)

cm86, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Additional Load is showing with value 0

### Expected behaviour

 there should be an option to hide the load if the power is below a limit or 0.

### Possible Solutions

ad an option for that... :)

### Mode

lite

### Context / Reason

like described

slipx06, 2024-12-17:
Why not just enable dynamic colour and use the off_threshold to grey our loads that are off.  I could also add an option to set the off colour and you can set it to transparent to hide the load instead of the default grey.

cm86, 2024-12-17:
As you can see it's already colored gray when the load is 0.

But as my washing machine and the tumble dryer aren't on all the day i would hide them when they are off...

[image]

thanks for your reply

slipx06, 2024-12-18:
OK, I've added an option to set the off colour for the additional load. When you set it to 'transparent' it will hide the load when <= off threshold

Off colour set to default grey
[image]

Off colour set to transparent
[image]

cm86, 2024-12-18:
Nice - Thanks.
i think i can test this in the next release?!

Mit freundlichen Grüßen
Chris

closed 2024-12-18 by slipx06

alphabravocharly72, 2024-12-20:
Nice feature, but cant find the transparent button. Or do I need to set the RGB colors?

slipx06, 2024-12-20:
@alphabravocharly72  you cant set it via the visual editor. Switch to the code editor and change the `off_colour` attribute in the load section  to `transparent`.

alphabravocharly72, 2024-12-20:
thx for the hint.
Added the off_colour manualy and it works perfectly.
Again thx for your feedback ;-)

commit a2304183 (2024-12-18): feat: adds `off_colour:` and `remaining_energy_to_shutdown:` closes #577 closes #554