slipx06/sunsynk-power-flow-card #579: Custom image for inverter

kawlsson, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

First off. Love the card!
Works perfectly after a little bit of tinkering. 

I have a Ferroamp energyhub 14 and would like to add it instead of the generic one. 
would that be possible to do?
Small cosmetic thing but still. 

### Expected behaviour

See my inverter on the card

### Possible Solutions

Allow custom images to be added to accommodate basically any inverter on the card. 

### Mode

Both

### Context / Reason

Who doesn’t like a little bit of bling

slipx06, 2024-12-26:
Hi. Is not possible to do this with the current code. I can add support for additional inverters if you provide an image and any custom inverter codes you need displayed with the inverter stats

lightzhuk, 2025-01-24:
please, is it possible to add error/warning statuses with solar2mqtt (easun)? maybe I can help with this? thank you very much

slipx06, 2025-01-24:
Yes I can add that. Please share the list of status codes

lightzhuk, 2025-01-25:
Please let me know if I can be of any help in this regard. Thank you.

lightzhuk, 2025-01-30:
[image]
EAsun 3.2kw

lightzhuk, 2025-01-30:
when everything is fine, status: "ok"

spoof54, 2025-03-11:
Hello, 
Not an inverter, but in my case and maybe for others a Linky image would be great :)

[image]
[image]

Thank you

slipx06, 2025-04-05:
@lightzhuk I've added the image, but I need more information for the status codes.  There should be one sensor that reads the inverter state. See code below

https://github.com/slipx06/sunsynk-power-flow-card/blob/main/src/inverters/brands/easun.ts

oriolism, 2025-10-23:
Could you please add the image for Generac PWRCell?

[image]