slipx06/sunsynk-power-flow-card #587: Please help! I want to convert the unit "W" to "A" at the position as in the picture, I hope to get help. Thank you very much.

thaibacgiang, 2024-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Please help! I want to convert the unit "W" to "A" at the position as in the picture, I hope to get help. Thank you very much.
[image]

### Expected behaviour

Please help! I want to convert the unit "W" to "A" at the position as in the picture, I hope to get help. Thank you very much. 
[image]

### Possible Solutions

Please help! I want to convert the unit "W" to "A" at the position as in the picture, I hope to get help. Thank you very much.

### Mode

lite

### Context / Reason

Please help! I want to convert the unit "W" to "A" at the position as in the picture, I hope to get help. Thank you very much.

slipx06, 2025-01-10:
Hi, That placeholder requires a sensor that provides power (W). The easiest fix is for you to create a fork and update as per your pull request

thaibacgiang, 2025-01-10:
> Hi, That placeholder requires a sensor that provides power (W). The easiest fix is for you to create a fork and update as per your pull request

How can I create an additional branch myself? Can the author please guide me? Thanks!

slipx06, 2025-01-10:
Have a look here on how to setup a dev environment 

https://slipx06.github.io/sunsynk-power-flow-card/contribute/devcontainer.html

You can then build your own version with the required change.