slipx06/sunsynk-power-flow-card #588: Custom element doesn't exist: sunsynk-power-flow-card

vdpollmza, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

i have installed Sunsynk-power-flow-card into HA as per the documentation. i can see the folder created on my HAS, but when on my dashboard, i try to add a new card, and choose Manual to add a custom card, when, and i want to add a custom type - custom:sunsynk-power-flow-card, i get the error that the custom element doesn't exist. please can you guide me as to what i am doing wrong. will buy you a coffee. thank you sir.
regards
Marc
[image]

### Steps To Reproduce

install as per steps, but cannot seem to add a custom card. when i google, it says that the Sunsynk-power-flow-card.js is not in the www folder, but when i go check, it is there.

### Expected behaviour

just want to be able to add a custom card for your solution to my dashboard, wanted to use your example card configuration, but for some reason, i cannot figure out how.

### Card Version

latest

### Home Assistant Version

2024.12.5

slipx06, 2025-01-02:
Hi. Did you use HACS to install the card? That’s the easiest way?

vdpollmza, 2025-01-03:
hello Slipx06, yes i did, that is what is confusing me.

let me remove it, reboot everything and try again.

By the way, do you have a coffee link that i can donate some funds to your work?

vdpollmza, 2025-01-03:
i removed it and rebooted and redid it, and it is there as a custom card now. whoop whoop. love your work.

thank you. i bought you a coffee.

regards
Marc

closed 2025-01-03 by vdpollmza