slipx06/sunsynk-power-flow-card #589: Displaying sensor details may get confused and refuse to open

kosi2801, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When I open a sensor detail view by clicking eg. the "Battery charged" sensor and then continue to the longer history via "Show more" the state if a detail view is currently displayed gets mixed up and prevents opening further sensor details on the Sunsynk card as long as the same procedure hasn't been done on a different dashboard to reset the state.
Also,in this corrupted state when a sensor detail view is displayed on a different dashboard and then closed again via the "X" the displayed dashboard is changed to the Sunsynk dashboard without actively clicking this dashboard.

My dashboard is set up as "Panel (single card)" only containing the Sunsynk card.

I assume that some status that holds the condition, if a detail sensor dialog is open is not properly reset when the dialog is not left via the "X" close button but via the "Show more" link.

### Steps To Reproduce

1. create a new dashboard with "Panel (single card)" config and add the Sunsynk card
2. set up the sunsynk card to display at least one clickable sensor
3. finish configuration and save the dashboard
4. click on one of the sensor links so that the detail dialog for this sensor is shown
5. click "Show more" in this dialog to go to the History dashboard for this sensor
6. click in the sidebar the link to the dashboard containing the Sunsynk card
7. try to click the same or another sensor
8. **!!!** nothing happens, no detail dialog is opened
9. click on _another_ dashboard with clickable sensors
10. click on a sensor in that dashboard to show the detail dialog
11. click on "X" to close the detail dialog again
12. **!!!** the displayed dashboard is suddenly reset to the one containing the Sunsynk card
13. now all sensor clicks work again

### Expected behaviour

In step 8 of the "Steps to reproduce" I would expect a sensor detail dialog to open and show the recent sensor history.
In step 12 of the "Steps to reproduce" I would expect to stay on that other dashboard and not implicitely be thrown to the dashboard where the Sunsynk card is shown.

### Card Version

6.2.0

### Home Assistant Version

2024.12.5

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  energy: "10200"
  show_daily: true
  count: 1
  auto_scale: true
  show_remaining_energy: true
  linear_gradient: true
  soc_end_of_charge: "100"
  shutdown_soc_offgrid: 5
  shutdown_soc: 20
  show_absolute: false
  remaining_energy_to_shutdown: true
  max_power: "25000"
  dynamic_colour: true
  animate: false
  colour: green
  charge-colour: lightgreen
solar:
  show_daily: true
  mppts: 2
  auto_scale: true
  pv1_name: West
  pv2_name: Ost
  display_mode: 2
  dynamic_colour: true
  max_power: 7500
  efficiency: 1
load:
  show_daily: true
  auto_scale: true
  load2_icon: mdi:water-boiler
  additional_loads: 3
  load1_name: Corsa
  load1_icon: mdi:car
  load2_name: Boiler
  load3_name: Heizung
  load3_icon: mdi:fire-circle
  dynamic_colour: true
  dynamic_icon: true
  label_daily_load: Verbrauch
  max_power: 11000
  aux_dynamic_colour: false
  off_threshold: 10
  off_colour: transparent
  path_threshold: 75
  invert_flow: false
  show_aux: false
  show_daily_aux: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  show_absolute: false
  label_daily_grid_buy: Import
  label_daily_grid_sell: Export
  auto_scale: true
  import_icon: mdi:transmission-tower-import
  export_icon: ""
  disconnected_icon: mdi:transmission-tower-off
  grid_name: " "
  max_power: 11000
  off_threshold: 50
entities:
  essential_power: none
  pv2_current_112: sensor.stp10_0_3se_40_xxx_pv_current_b
  pv1_power_186: sensor.stp10_0_3se_40_xxx_pv_power_a
  pv2_power_187: sensor.stp10_0_3se_40_xxx_pv_power_b
  day_pv_energy_108: sensor.stp10_0_3se_40_xxx_daily_yield
  pv1_current_110: sensor.stp10_0_3se_40_xxx_pv_current_a
  environment_temp: sensor.node_1_input_1
  total_pv_generation: sensor.stp10_0_3se_40_xxx_total_yield
  pv_total: sensor.stp10_0_3se_40_xxx_pv_power
  battery_soc_184: sensor.stp10_0_3se_40_xxx_battery_soc_total
  battery_soh: sensor.stp10_0_3se_40_xxx_battery_capacity_total
  battery_current_191: sensor.stp10_0_3se_40_xxx_battery_current_a
  battery_temp_182: sensor.stp10_0_3se_40_xxx_battery_temp_a
  battery_status: sensor.stp10_0_3se_40_xxx_battery_status_operating_mode
  grid_ct_power_172: sensor.stp10_0_3se_40_xxx_metering_active_power_draw_l1
  grid_ct_power_L2: sensor.stp10_0_3se_40_xxx_metering_active_power_draw_l2
  grid_ct_power_L3: sensor.stp10_0_3se_40_xxx_metering_active_power_draw_l3
  inverter_status_59: sensor.stp10_0_3se_40_xxx_status
  inverter_current_164: sensor.stp10_0_3se_40_xxx_metering_current_l1
  load_frequency_192: sensor.stp10_0_3se_40_xxx_metering_frequency
  inverter_current_L2: sensor.stp10_0_3se_40_xxx_metering_current_l2
  inverter_current_L3: sensor.stp10_0_3se_40_xxx_metering_current_l3
  inverter_power_175: sensor.stp10_0_3se_40_xxx_metering_power_supplied
  grid_power_169: sensor.stp10_0_3se_40_xxx_metering_power_absorbed
  essential_load1: sensor.myenergi_garage_power_charging
  essential_load2: sensor.shellyplusplugs_aaaaaabbbbbb_switch_0_power
  essential_load3: sensor.shellyplusplugs_bbbbbbcccccc_switch_0_power
  remaining_solar: sensor.energy_production_today_remaining
  battery_voltage_183: sensor.stp10_0_3se_40_xxx_battery_voltage_a
  pv1_voltage_109: sensor.stp10_0_3se_40_xxx_pv_voltage_a
  pv2_voltage_111: sensor.stp10_0_3se_40_xxx_pv_voltage_b
  battery_power_190: sensor.batteriestrom_stp_w_absolut_helper
  battery_rated_capacity: "10200"
  grid_connected_status_194: sensor.stp10_0_3se_40_xxx_grid_relay_status
  day_battery_charge_70: sensor.batterieladung_stp_kwh_daily_helper
  day_battery_discharge_71: sensor.batterienutzung_stp_kwh_daily_helper
  day_grid_export_77: sensor.netzeinspeisung_stp_wh_daily_helper
  day_grid_import_76: sensor.netzverbrauch_stp_wh_daily_helper
  day_load_energy_84: sensor.hausverbrauch_stp_wh_calc_daily_helper
  load_power_L3: sensor.stromquellen_stp_l3_gesamt_aktuell
  load_power_L1: sensor.stromquellen_stp_l1_gesamt_aktuell
  load_power_L2: sensor.stromquellen_stp_l2_gesamt_aktuell
  essential_load1_extra: sensor.psa_battery_level
  essential_load2_extra: sensor.node_1_input_2
  essential_load3_extra: sensor.node_1_input_5
large_font: true
wide: true
show_battery: true
show_grid: true
dynamic_line_width: true
inverter:
  three_phase: true
  model: smasolar
  modern: false
  autarky: energy
  auto_scale: true
```

kosi2801, 2025-01-08:
After updating to HA 2025.1 the behavior has changed and when clicking "Show more" no details are displayed but instead the dialog disappears and the whole Sunsynk dashboard seems to flash once (being reloaded?).

bundi78, 2025-01-08:
Same for me. Clicking on "Show more" just closes the dialog but does not open the history-view. I'm running HA 2025.1.1

molikk, 2025-01-09:
@slipx06 
Looks like removing  history.back();  from _dispatchMoreInfoEvent in src/helpers/utils.ts solves the issue

closed 2025-01-10 by slipx06

kosi2801, 2025-01-10:
Thank you very much for the fast turnaround 😀
Checked and seems to work correctly now.

commit 41539b22 (2025-01-10): fix: show more in history chart not working closes  #589