slipx06/sunsynk-power-flow-card #590: Off-Grid mode display

Sn0w82, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

At this point if the grid is turned off or disconnects due to an outage, the card only stops showing the grid details and animation.

### Expected behaviour

My suggestion will be that once grid is detected as not available card to display an X on the grid flow and display a off-grid small text message next to the X so it’s easier to spot and more nice as design.

added a picture as a reference for my proposal.
[image]

### Mode

Both

### Context / Reason

Design and executive look improvment.

slipx06, 2025-01-10:
Hi. You can set both the colour and the icon to be displayed when the grid is disconnected,  See

`disconnected_icon:`
`grid_off_colour:`

[image]

Sn0w82, 2025-01-10:
Didn’t saw  that. Thanks. Is doing the job.
Btw, everywhere i have the card flow card in swipe card mode, the grid import/export/disconnect pictures based on status is displayed in the top upper left corner. Picture attached.

where the flow card is deployed without any other cards in config, all is good and working as expected so i thibk it’s a bug from the swipe mix or something.
[image]

slipx06, 2025-01-11:
Can you share your config and the custom cards you are using for the swiping?

Sn0w82, 2025-01-11:
Config is here. 
https://pastecode.io/s/56fbih0e

Cards used are:
- custom:swipe-card --- as main type card
- custom:sunsynk-power-flow-card --- for 1st card in the swipe
- vertical-stack --- for 2nd card in the swipe

slipx06, 2025-01-11:
What browser are you using? There are some issues with Safari and foreign elements?
Try downloading the latest version 6.2.2. Its working for me in Chrome and Firefox

[image]

Sn0w82, 2025-01-11:
I use chrome on my workstation and ios app on my phone and wife phone. All 3 show the same behavior in swipe card configuration and with different icons for grid status:

      disconnected_icon: mdi:transmission-tower-off
      export_icon: mdi:transmission-tower-import
      import_icon: mdi:transmission-tower-export

attaching also video with above icons not configured — all good. 
Another one with above configured inside card config and issue being present.

https://github.com/user-attachments/assets/e39d93ad-2c22-47d3-b00c-87908451486c

https://github.com/user-attachments/assets/4934d259-933f-4617-ad32-50924d71195f

Sn0w82, 2025-01-13:
Found a possible way for you to reproduce the issue it on your side:
Looks like when in swipe mode card, what you put as navigation target for grid icon acts as a solar navigation target too — that’s why grid icon is shown in same place with solar icon — as they basically take you to same target (the one setup for grid icon) even if you have different targets between solar and grid icons.