slipx06/sunsynk-power-flow-card #595: No Flow Animation on HA iOS App

limatango79, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Power Flow Animation available on Desktop Browser, but not on HA iOS App as used to be before.

### Expected behaviour

Energy Flow Animation in iOS App

### Card Version

6.4.0

### Home Assistant Version

2025.1.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  shutdown_soc: "10"
  show_daily: false
  auto_scale: true
  show_absolute: false
  invert_power: true
  colour:
    - 85
    - 134
    - 180
  charge_colour:
    - 129
    - 134
    - 156
  dynamic_colour: false
  linear_gradient: false
  hide_soc: true
  animation_speed: 1
  max_power: 1
  show_remaining_energy: false
  animate: false
solar:
  show_daily: true
  mppts: 4
  pv1_name: Nord
  pv2_name: Süd
  pv3_name: West
  pv4_name: Ost
  auto_scale: true
  display_mode: 1
  dynamic_colour: true
  animation_speed: 1
  max_power: "26520"
  colour:
    - 208
    - 167
    - 78
  pv1_max_power: 8580
  pv2_max_power: 8580
  pv3_max_power: 4680
  pv4_max_power: 4680
  efficiency: 2
load:
  show_daily: false
  auto_scale: true
  dynamic_icon: false
  additional_loads: 2
  essential_name: HAUS
  load1_name: EV
  load1_icon: mdi:ev-station
  load2_name: WP
  load2_icon: mdi:heat-pump
  colour:
    - 203
    - 93
    - 93
  dynamic_colour: false
  show_aux: false
  animation_speed: 2
  max_power: 1
  invert_load: false
grid:
  show_daily_buy: false
  show_daily_sell: false
  show_nonessential: false
  auto_scale: true
  invert_grid: true
  grid_name: NETZ
  show_absolute: false
  animation_speed: 1
  export_colour:
    - 63
    - 146
    - 88
entities:
  battery_soc_184: sensor.bat_soc
  battery_power_190: sensor.bat_leistung
  day_pv_energy_108: sensor.pv_heute
  pv1_power_186: sensor.pv_nord_leistung
  pv2_power_187: sensor.pv_sud_leistung
  pv3_power_188: sensor.pv_west_leistung
  pv4_power_189: sensor.pv_ost_leistung
  essential_power: sensor.verbrauch_knx_haus_gesamt_leistung
  essential_load1: sensor.openwb_ladeleistung
  essential_load2: sensor.verbrauch_knx_wp_gesamt_leistung
  load_frequency_192: sensor.ac_netzfrequenz
  inverter_power_175: sensor.ac_wr_wirkleistung_total
  grid_power_169: sensor.ac_bezug_einspeisung
  solar_sell_247: "on"
  day_load_energy_84: sensor.verbrauch_heute
  grid_ct_power_172: sensor.ac_bezug_einspeisung
large_font: true
panel_mode: true
show_battery: true
show_grid: true
dynamic_line_width: false
inverter:
  three_phase: false
  model: sofar
  modern: true
  autarky: "no"
  auto_scale: true
  colour:
    - 158
    - 139
    - 167
decimal_places: 1
decimal_places_energy: 0
wide: false
```

slipx06, 2025-01-12:
Strange. Its working fine on my side and there have been no changes to any of the animation code. Maybe try reset your frontend cache in the HA app.

limatango79, 2025-01-12:
I've resetted the frontend cache several times, deinstalled the app and installed it again, tried different card styles and setups.
But no change in behavior.

slipx06, 2025-01-12:
Does it work in Safari?

limatango79, 2025-01-12:
It does not work neither on Safari (iOS) nor on Brave (iOS). I'm on iOS 18.2.1

I've created a basic flow card with Inverter and Load only, but the same result. Flow animation on Desktop (MacOS) and no animation on iOS.

BUT: IT WORKS ON A SECOND iPHONE. Which setting in the iOS system could influence this behavior?

limatango79, 2025-01-12:
OK.

The solution is: iOS -> Settings -> Accessibility -> Motion -> "Autoplay Animated Images" must be activated.

Sorry... I did not know that.

closed 2025-01-12 by slipx06