slipx06/sunsynk-power-flow-card #596: Solax inverter status enhancement

Reinhard-M, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Status of the inverter is not shown, just "Status".

### Expected behaviour

Support additional status values.

### Possible Solutions

        this.statusGroups = {
            normal: {
//                states: ["normal mode"],
                states: ["2", "normal", "normalmode", "normal mode"],
                color: "green",
                message: Et("common.normal")
            },
            standby: {
//                states: ["waiting", "checking", "idle mode", "standby"],
                states: ["0", "waiting", "waitmode", "9", "idle", "idlemode", "idle mode", "10", "standby", "standbymode"],
                color: "blue",
                message: Et("common.standby")
            },
            selftest: {
//                states: ["self test"],
                states: ["8", "self testing", "selftest", "self test"],
                color: "yellow",
                message: Et("common.selftest")
            },
            offgrid: {
//                states: ["off-grid waiting", "off-grid", "eps mode"],
                states: ["6", "off-grid waiting", "epscheckmode", "7", "off-grid", "epsmode", "eps mode"],
                color: "green",
                message: Et("common.offgrid")
            },
            fault: {
//                states: ["permanent fault mode", "fault"],
                states: ["4", "permanent fault", "permanentfaultmode", "permanent fault mode", "3", "fault", "faultmode"],
                color: "red",
                message: Et("common.fault")
            },
            check: {
//                states: ["checking", "update mode", "eps check mode"],
                states: ["1", "checking", "checkmode", "5", "update", "updatemode", "update mode", "eps check mode"],
                color: "orange",
                message: Et("common.check")
            }
        },

### Mode

lite

### Context / Reason

I'm using the "SolaxModbusGateway" for my Solax inverter. Unfortunately the messages of this gateway are not covered by the actual sunsync card. I've adapted the respective js file and added the changes to "Possible Solutions". All of the new values are tested and are working well on my system.
Additional Information:
I've added the states from the original Solax  and the SolaxModbusGateway documentation. The first 2 states parameter are always the original Solax parameter (e.g "2", "normal"), the third one (..., "normalmode", ...) comes from the SolaxModbusGateway documentation. Everything else is as it was. Except "checking". It was mentioned twice, once assigned to "standby" and once assigned to "check". For me I've removed it from "standby".

closed 2025-01-12 by slipx06

Reinhard-M, 2025-01-13:
@slipx06: Sorry, but you have forgotten to change the "normal state" line. Everything else is working as requested, thanks a lot.

slipx06, 2025-01-13:
I'll push a fix now

Reinhard-M, 2025-01-13:
Just checked with v6.4.3, working well, thanks for the very fast response :-)

commit 418a78a6 (2025-01-12): fix: update missing Solax inverter status messages closes #596