slipx06/sunsynk-power-flow-card #600: autarky and ratio score  very often wrong (deviations of 5-80%)

justinh998, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hello, I have noticed that the autark / score values are very often not correct and I sometimes have up to 80% deviation from the real value.

Here is an example for the autark: 

[image]

In the documentation it says that the autarky is Calculated by :

 Home Production / Home Consumption.

My home production in this case is 1009W and the home consumption is 1936 watts and 1009/1936 = 0.521178(rounded) but even if I round to the nearest 10 I will never get 0.35(35%). 

It is the same with the ratio.
If I only get power from solar and the battery is neither running nor discharging, the deviations are still okay (usually they are "only" 5-15%) but as soon as the battery is added, it is sometimes up to 80%.

Does anyone have any idea what is going wrong here?

### Expected behaviour

Both percentages are calculated correctly.

### Card Version

6.4.3

### Home Assistant Version

2025.1.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  energy: 2000
  shutdown_soc: 10
  show_daily: true
  count: 1
  max_power: 800
  soc_end_of_charge: 100
  auto_scale: true
  dynamic_colour: true
  linear_gradient: true
  animate: true
  show_remaining_energy: true
  remaining_energy_to_shutdown: true
  show_absolute: false
solar:
  show_daily: true
  mppts: 2
  pv1_name: Solarplatte 1+2
  pv2_name: Solarplatte 3
  auto_scale: true
  dynamic_colour: true
  max_power: 1200
  pv1_max_power: 800
  pv2_max_power: 400
load:
  show_daily: true
  dynamic_colour: true
  dynamic_icon: true
  essential_name: Haus
  load1_name: Boiler unten
  additional_loads: 1
  load1_icon: mdi:water-boiler
  max_power: 2000
  auto_scale: true
  label_daily_load: Heutiger Verbrauch
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  grid_name: Stromnetz
  invert_grid: false
  auto_scale: true
  show_absolute: false
  max_power: 2000
  label_daily_grid_buy: "Strom vom Netz heute "
  label_daily_grid_sell: "Strom ans Netz heute "
entities:
  day_battery_charge_70: sensor.noah_2000_ladung_heute
  battery_soc_184: sensor.noah_2000_soc
  battery_power_190: sensor.noah_2000_lade_entladesensor
  grid_ct_power_172: sensor.bitshake_smartmeterreader_ddz285_power
  essential_power: sensor.aktueller_stromverbrauch
  day_pv_energy_108: sensor.solarpower_heute
  pv1_power_186: sensor.noah_2000_solar_power
  pv2_power_187: sensor.wechselrichter_port_3_dc_leistung
  essential_load1: sensor.boiler_unten_derzeitiger_verbrauch
  battery_temp_182: sensor.noah_2000_bat0_temperature
  day_battery_discharge_71: sensor.noah_2000_heutige_entladung
  day_grid_import_76: sensor.strom_vom_netz_heute
  day_grid_export_77: sensor.strom_ins_netz_heute
  day_load_energy_84: sensor.stromverbrauch_haus_heute
large_font: true
show_battery: true
show_grid: true
dynamic_line_width: true
decimal_places: 3
decimal_places_energy: 3
inverter:
  auto_scale: true
  autarky: power
  modern: true
wide: false
```

### Relevant log output

```shell

```

slipx06, 2025-01-18:
Hi I cant see why that is not working and you can see the logic in the code below which the card uses.

In your case production is only the totalPV  value  (calculated by the card logic) as battery power is 0 and you don't use AUX
My suggestion is that you create some template sensors and substitute for battery, pv_total and essential to test and try eliminate the issue.

The sensors you need to create are:

```
essential_power: test.sensor1
battery_power_190: test.sensor2
pv_total: test,sensor3 <-----you can set your own sensor for the total pv which overides the card logic for adding pv1 and pv2

```

```
const productionPower =
            totalPV +
            Utils.toNum(`${(config.battery.invert_flow === true ? batteryPowerTotal < 0 : batteryPowerTotal > 0) ? Math.abs(batteryPowerTotal) : 0}`) +
            Utils.toNum(`${auxPower < 0 ? auxPower * -1 : 0}`);
```

```
const consumptionPower =
            essentialPower +
            Math.max(nonessentialPower, 0) +
            Utils.toNum(`${auxPower > 0 ? auxPower : 0}`) +
            Utils.toNum(`${(config.battery.invert_flow === true ? batteryPowerTotal > 0 : batteryPowerTotal < 0) ? Math.abs(batteryPowerTotal) : 0}`);
```

```
const autarkyPower = consumptionPower != 0 ? Math.max(Math.min(Math.round((productionPower * 100) / consumptionPower), 100), 0) : 0;
        const ratioPower = productionPower != 0 ? Math.max(Math.min(Math.round((consumptionPower * 100) / productionPower), 100), 0) : 0;
```

slipx06, 2025-01-18:
Here is a dev version with logging enabled . Add a new dashboard resource and unzip and copy the js file into your  www folder.  You can then add a new card of type: custom:sunsynk-power-flow-card-dev

[sunsynk-power-flow-card-dev.zip](https://github.com/user-attachments/files/18465400/sunsynk-power-flow-card-dev.zip)

[image]

[image]

If you open your browser console logs you can see the autarky data

[image]

slipx06, 2025-02-05:
I'm closing this as I've not received any feedback and cant see any issue with the code. If you can provide more feedback  and 
the logs from the dev version I posted we can re-open.

closed 2025-02-05 by slipx06