slipx06/sunsynk-power-flow-card #610: PV efficiency bug

daimonkor, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Please check why PV1 efficiency always 100 %

  
[image]

[image]

[image]

### Expected behaviour

Calculate power and rate percent fron max_power

### Card Version

6.5.1

### Home Assistant Version

2025.1.4

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
large_font: true
battery:
  shutdown_soc: number.ss_battery_shutdown_capacity
  invert_power: false
  colour: var(--energy-battery-in-color)
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  linear_gradient: true
  animate: true
  show_remaining_energy: true
  max_power: 630
solar:
  colour: var(--energy-solar-color)
  show_daily: true
  mppts: 1
  auto_scale: true
  dynamic_colour: true
  pv1_name: Юг
  efficiency: 3
  max_power: 4468
entities:
  use_timer_248: switch.ss_use_timer
  priority_load_243: switch.ss_priority_load
  inverter_voltage_154: sensor.ss_grid_voltage
  inverter_load_freq_192: sensor.ss_grid_frequency
  inverter_current_164: sensor.ss_inverter_current
  grid_connected_status_194: binary_sensor.ss_grid_connected
  inverter_status_59: sensor.ss_overall_state
  batchargeday_70: sensor.ss_day_battery_charge
  batdischargeday_71: sensor.ss_day_battery_discharge
  battery_voltage_183: sensor.ss_battery_voltage
  battery_soc_184: sensor.ss_battery_soc
  battery_power_190: sensor.ss_battery_power
  battery_current_191: sensor.ss_battery_current
  battery_temp_182: sensor.ss_battery_temperature
  grid_power_169: sensor.ss_grid_power
  grid_buy_day_76: sensor.ss_day_grid_import
  grid_sell_day_77: sensor.ss_day_grid_export
  loadday_84: sensor.ss_day_load_energy
  load_frequency_192: sensor.ss_load_frequency
  essential_power: sensor.ph1800_real_load_power
  nonessential_power: sensor.ss_non_essential_power
  aux_power_166: sensor.ss_aux_power
  solarday_108: sensor.ss_day_pv_energy
  pv1_power_186: sensor.ss_pv1_power
  pv1_voltage_109: sensor.ss_pv1_voltage
  pv1_current_110: sensor.ss_pv1_current
  radiator_temp_91: sensor.ss_radiator_temperature
  dc_transformer_temp_90: sensor.ss_dc_transformer_temperature
  prog1_time: select.ss_prog1_time
  battery_rated_capacity: sensor.ss_battery_rated_capacity
  grid_ct_power_172: sensor.ph1800_abs_real_grid_power
  inverter_power_175: sensor.ss_inverter_power
  day_grid_import_76: sensor.ss_range_ph1800_real_accumulated_buy_grid_power
  day_pv_energy_108: sensor.ss_range_custom_accumulated_pv_power
  day_battery_charge_70: sensor.ss_range_ph1800_real_accumulated_battery_charger_power
  day_battery_discharge_71: sensor.ss_range_ph1800_real_accumulated_discharge_battery_power
  day_load_energy_84: sensor.ss_range_ph1800_real_accumulated_load_power
show_battery: true
show_grid: true
inverter:
  auto_scale: true
  modern: false
  model: deye
  autarky: energy
  show_daily: true
load:
  show_daily: true
  auto_scale: true
  dynamic_colour: true
  dynamic_icon: true
  max_colour: true
grid:
  show_daily_buy: true
view_layout:
  position: sidebar
```

### Relevant log output

```shell

```

slipx06, 2025-01-27:
Hi you need to define the max_power for PV1. Add this to your config

`pv1_max_power: 4468`

I know its redundant as you only have one string and it should be the same as max_power but for now thats how it works 😄

daimonkor, 2025-01-27:
@slipx06 thanks it works) magic decision), sorry for delay

closed 2025-01-27 by slipx06