slipx06/sunsynk-power-flow-card #615: Icons dissapear

geronet1, 2025-02-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Battery and Home icons dissapear after some time

[image]

Fix: Reload with browser or close and re-open the app 

### Expected behaviour

n.a.

### Card Version

v6.5.1

### Home Assistant Version

2025.1.4

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  energy: 21300
  shutdown_soc: 10
  show_daily: true
  show_remaining_energy: true
  animate: false
  linear_gradient: true
  hide_soc: true
  invert_power: false
  auto_scale: false
  show_absolute: false
  dynamic_colour: true
  max_power: 9600
  charge_colour:
    - 43
    - 255
    - 0
  colour:
    - 255
    - 255
    - 255
  remaining_energy_to_shutdown: true
solar:
  show_daily: true
  mppts: 4
  pv1_name: Nord
  pv2_name: Süd
  pv3_name: Garage
  pv4_name: HMS
  dynamic_colour: true
  auto_scale: false
  display_mode: 3
  pv2_max_power: 7020
  pv3_max_power: 12480
  pv4_max_power: 1560
  pv1_max_power: 7040
  max_power: "28100"
  colour:
    - 20
    - 162
    - 2
  navigate: /lovelace/2
load:
  show_daily: true
  dynamic_icon: true
  colour:
    - 186
    - 186
    - 186
  dynamic_colour: true
  additional_loads: 2
  load1_icon: mdi:solar-power-variant
  show_aux: false
  aux_type: ""
  load1_name: VE Süd
  show_daily_aux: false
  invert_load: false
  load2_name: VE Nord
  load2_icon: mdi:solar-power-variant
  invert_flow: false
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  invert_grid: false
  show_absolute: false
  label_daily_grid_sell: Lieferung
  label_daily_grid_buy: Bezug
  grid_name: Netz
  grid_off_colour:
    - 255
    - 0
    - 0
  export_colour:
    - 0
    - 255
    - 4
  colour:
    - 255
    - 136
    - 122
  no_grid_colour:
    - 30
    - 134
    - 9
  additional_loads: 0
  nonessential_icon: ""
  off_threshold: 20
  max_power: 10000
  auto_scale: true
  import_icon: mdi:transmission-tower-import
  export_icon: mdi:transmission-tower-export
  disconnected_icon: mdi:transmission-tower-off
  load1_icon: ""
entities:
  use_timer_248: switch.deye_time_of_use
  priority_load_243: switch.deye_toggle_priority_load
  inverter_voltage_154: sensor.deye_grid_voltage_l1
  load_frequency_192: sensor.deye_load_frequency
  inverter_current_164: sensor.deye_grid_current_l1
  inverter_power_175: sensor.deye_inverter_power
  grid_connected_status_194: binary_sensor.deye_ac_grid_relay
  inverter_status_59: sensor.deye_running_status
  day_battery_charge_70: sensor.deye_daily_charge_of_the_battery
  day_battery_discharge_71: sensor.deye_daily_discharge_of_the_battery
  battery_voltage_183: sensor.deye_battery_voltage
  battery_soc_184: sensor.deye_battery_soc
  battery_power_190: sensor.deye_battery_power
  battery_current_191: sensor.deye_battery_current
  grid_power_169: sensor.deye_grid_power
  day_grid_import_76: sensor.deye_daily_energy_bought
  day_grid_export_77: sensor.deye_daily_energy_sold
  grid_ct_power_172: sensor.deye_external_ct_l1_power
  day_load_energy_84: sensor.verbrauch_import_daily
  essential_power: sensor.smlreader_verbrauch_power
  day_pv_energy_108: sensor.deye_daily_production
  pv1_power_186: sensor.deye_pv1_power
  pv2_power_187: sensor.deye_pv2_power
  pv1_voltage_109: sensor.deye_pv1_voltage
  pv1_current_110: sensor.deye_pv1_current
  pv2_voltage_111: sensor.deye_pv2_voltage
  pv2_current_112: sensor.deye_pv2_current
  pv3_power_188: sensor.symo_10_0_3_m_leistung_ac
  pv4_power_189: sensor.d_p_ac
  grid_ct_power_L2: sensor.deye_external_ct_l2_power
  grid_ct_power_L3: sensor.deye_external_ct_l3_power
  inverter_voltage_L2: sensor.deye_grid_voltage_l2
  inverter_voltage_L3: sensor.deye_grid_voltage_l3
  inverter_current_L2: sensor.deye_grid_current_l2
  inverter_current_L3: sensor.deye_grid_current_l3
  radiator_temp_91: sensor.deye_heat_sink_temperature
  battery_temp_182: sensor.deye_battery_temperature
  grid_ct_power_total: sensor.smlreader_haus_power
  energy_cost_buy: sensor.am_brucklesbach_1_strompreis
  solar_sell_247: switch.deye_solar_sell
  load_power_L1: sensor.smlreader_verbrauch_power_l1
  max_sell_power: number.deye_max_solar_sell_power
  total_pv_generation: sensor.deye_total_pv_production
  prog1_time: sensor.deye_time_slot_1
  prog2_time: sensor.deye_time_slot_2
  prog3_time: sensor.deye_time_slot_3
  prog4_time: sensor.deye_time_slot_4
  prog5_time: sensor.deye_time_slot_5
  prog6_time: sensor.deye_time_slot_6
  prog1_capacity: number.deye_time_point_1_capacity
  prog2_capacity: number.deye_time_point_2_capacity
  prog3_capacity: number.deye_time_point_3_capacity
  prog4_capacity: number.deye_time_point_4_capacity
  prog5_capacity: number.deye_time_point_5_capacity
  prog6_capacity: number.deye_time_point_6_capacity
  prog1_charge: switch.deye_time_point_1_charge_enable
  prog2_charge: switch.deye_time_point_2_charge_enable
  prog3_charge: switch.deye_time_point_3_charge_enable
  prog4_charge: switch.deye_time_point_4_charge_enable
  prog5_charge: switch.deye_time_point_5_charge_enable
  prog6_charge: switch.deye_time_point_6_charge_enable
  load_power_L2: sensor.smlreader_verbrauch_power_l2
  load_power_L3: sensor.smlreader_verbrauch_power_l3
  essential_load1: sensor.smlreader_pv_power
  essential_load1_extra: sensor.pv_export_daily_kwh
  pv4_voltage_115: sensor.d_u_ac
  pv4_current_116: sensor.d_i_ac
  pv5_power: sensor.symo_10_0_3_m_leistung_ac
  pv3_current_114: sensor.symo_10_i_ac
  essential_load2: sensor.ve_nord_p_ac
show_battery: true
show_grid: true
inverter:
  three_phase: true
  model: deye
  auto_scale: false
  modern: true
  autarky: energy
  colour:
    - 133
    - 133
    - 133
large_font: true
panel_mode: true
dynamic_line_width: true
max_line_width: 6
min_line_width: 2
view_layout:
  position: main
card_width: 90%
wide: false
card_height: "900"
```

### Relevant log output

```shell

```

slipx06, 2025-02-03:
Hi. It looks to be related to the linear gradients that are applied when using the `dynamic_colour: true` setting. I don't know what is causing this but you can set `dynamic_colour: false` and it should stop this from happening. For the battery you also need to set `linear_gradient: false`

@Pho3niX90 if you have some time please can you help me with this. I can reproduce this in the editor by changing a random display setting. When I close the editor the icons are there but some users are reporting that they disappear after a while. 

[image]

slipx06, 2025-02-03:
Thanks. This behavior has been present for some time. I went back to v4.40.1 and can reproduce this.

baskarz, 2025-02-10:
I have been experiencing this too.
Ever since I updated home assistant to latest version. 
I also updated sunsynk to latest version.
As you can see, the home and battery icons are missing, and it is random. Sometimes it's fine.
I will try the fix as mentioned by slipx06 in the meantime.

[image]

Pho3niX90, 2025-02-10:
Yes mine was also more noticeable since the latest 2025.1/2 release, the PR should fix this, as it seems there is a change on how they purge the dom, might even be a bump in the LIT version. 

The PR ensures all SVG's have unique ID's now

closed 2025-02-10 by slipx06

geronet1, 2025-02-12:
[image]

Now the two mppts frames don't work

geronet1, 2025-02-12:
Ok thanks that helped

commit d4c26bb4 (2025-02-10): fix(svg): ensure unique gradient IDs to prevent missing fills, fixes #615