slipx06/sunsynk-power-flow-card #618: Manual edits to configuration are not retained; revert to prior configuration

peteDDD, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

When manually editing power-flow-card configuration using "show code editor" editor, pasted and saved code reverts to prior settings.  No changes made to through the editor "stick"

### Steps To Reproduce

1. Open "show code editor" screen from Sunsynk Power Flow Card configuration screen.
2. Add valid  line to the code shown there.
3. Click save.
4. Click done.
5. Return to editor following prior 4 steps.
6. Observe that changes made are no longer there.

### Expected behaviour

Changes made in editor override older code and remain after saving.

### Card Version

0.2.4

### Home Assistant Version

Core 2025.1.4,  Supervisor 2024.12.3,  Operating System 14.2,  Frontend 20250109.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
battery:
  energy: 16384
  shutdown_soc: 20
  show_daily: true
  count: 1
  max_power: 8000
  shutdown_soc_offgrid: 5
  soc_end_of_charge: 100
  remaining_energy_to_shutdown: true
  show_remaining_energy: true
  invert_flow: true
solar:
  show_daily: true
  mppts: 2
  pv1_name: PV1
  pv2_name: PV2
load:
  show_daily: true
  load1_icon: ""
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: false
  grid_name: Naturgy
  disconnected_icon: mdi:power-plug-off-outline
entities:
  inverter_voltage_154: sensor.bms_inv_inverter_voltage_a
  inverter_current_164: sensor.bms_inv_inverter_load_current_a
  inverter_power_175: srne_current_load_Power_a
  load_frequency_192: sensor.bms_inv_inverter_frequency
  grid_connected_status_194: binary_sensor.inverter_grid_connected_status
  inverter_status_59: sensor.bms_inv_charge_state
  day_battery_charge_70: sensor.bms_inv_battery_charge_today
  day_battery_discharge_71: sensor.bms_inv_battery_discharge_today
  battery_voltage_183: sensor.bms_inv_battery_voltage
  battery_soc_184: sensor.bms_inv_battery_soc
  battery_power_190: sensor.bms_inv_bms1_power
  battery_current_191: sensor.bms_inv_bms1_current
  grid_power_169: sensor.sunsynk_grid_power
  day_grid_import_76: sensor.bms_inv_line_power_consumption_of_the_day
  grid_ct_power_172: sensor.bms_inv_grid_a_power
  grid_ct_power_L2: sensor.bms_inv_grid_b_power
  grid_voltage: sensor.average_grid_voltage_phase_a_b
  day_load_energy_84: sensor.bms_inv_load_power_consumption_of_the_day
  essential_power: none
  nonessential_power: none
  day_pv_energy_108: sensor.sunsynk_day_pv_energy
  pv1_power_186: sensor.sunsynk_pv1_power
  pv2_power_187: sensor.sunsynk_pv2_power
  pv1_voltage_109: sensor.sunsynk_pv1_voltage
  pv1_current_110: sensor.sunsynk_pv1_current
  pv2_voltage_111: sensor.sunsynk_pv2_voltage
  pv2_current_112: sensor.sunsynk_pv2_current
  battery_rated_capacity: sensor.bms_inv_bms1_actual_battery_capacity
  battery_soh: sensor.bms_inv_bms1_state_of_health
  inverter_voltage_L2: sensor.bms_inv_inverter_voltage_b
  inverter_current_L2: sensor.bms_inv_inverter_load_current_B
  dc_transformer_temp_90: sensor.bms_inv_dc_to_dc_temperature
  radiator_temp_91: sensor.bms_inv_dc_to_ac_temperature
  load_power_L1: sensor.bms_inv_current_load_power_a
  load_power_L2: sensor.bms_inv_current_load_power_b
show_battery: true
inverter:
  three_phase: true
  autarky: energy
decimal_places_energy: 1
decimal_places: 2
dynamic_line_width: true
large_font: true
show_grid: true
```

### Relevant log output

```shell

```

slipx06, 2025-02-04:
Hi. Can you give a specific example of which setting(s) is being overridden as I cannot reproduce this? If I change something using the code editor is is displayed in the visual editor. Are you sure that you don't have the card editor open on two different browsers or PC's at the same time that can cause a conflict?

peteDDD, 2025-02-04:
@slipx06  That must have been the issue.  I closed down all browser windows and that seems to have fixed it.  I thought I had already tried that but a reboot of the PC seems to have helped as well... Thank you.

closed 2025-02-04 by slipx06