slipx06/sunsynk-power-flow-card #619: only home consumption value inside the house symbol

wifi75, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

currently we have the LOAD consumption of each device and then the total consumption of the house.

### Expected behaviour

you could have a sensor that generates the consumption of the house only, then you would only have to subtract the consumption of each single LOAD to generate the consumption of the house only. this value to be inserted as in the attached photo inside the house symbol.

### Possible Solutions

[image]

### Mode

lite

### Context / Reason

you could have a sensor that generates the consumption of the house only, then you would only have to subtract the consumption of each single LOAD to generate the consumption of the house only. this value to be inserted as in the attached photo inside the house symbol.

slipx06, 2025-02-11:
That icon changes so it wont be a good idea to place text there. As an alternative you can create a template sensor that subtracts all your measured loads from your essential load and call it "house load". You can then add that as an additional essential load.

wifi75, 2025-02-11:
Ok I can try!