slipx06/sunsynk-power-flow-card #620: Multiple inverters

rsmalexr, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Is it possible to add multiple inverts for upcoming additions? Use case master and slave running 2 Deye 40kw 

### Expected behaviour

U

### Mode

full

### Context / Reason

As above

ugath, 2025-03-13:
I have 2 microinverters and would also be interested.

rsmalexr, 2025-10-03:
@slipx06 i would be super into sending some coffee money for this addtion 👍  I have 2 sites with 3 ohase string of either 2 or three inverters the ability to hold all that data in one card withou haveing to make helpers etc would be epic