slipx06/sunsynk-power-flow-card #626: Include "last update" on card

BlythMeister, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Unable to know how current the data is

### Expected behaviour

Visual to show data accurate as of x minutes ago

### Possible Solutions

Small "last updated" in bottom corner

### Mode

Both

### Context / Reason

Had a situation where my sensor stopped updating so was looking at old data and unaware.

BlythMeister, 2025-02-15:
Hi, I didn't think about doing think kind of thing.

I'm wondering if it can be achieved as a custom markdown footer onto the card rather than another card stuck underneath?

slipx06, 2025-02-15:
You can use a stack in card to group multiple cards together

BlythMeister, 2025-02-15:
I feel like such a n00b.
Your right and I got this working.

closed 2025-02-15 by BlythMeister