slipx06/sunsynk-power-flow-card #630: Firefox on iPad not generating proper output

makcorner, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Simply as in subject, here is how it looks like:

[image]

I could observe it only on Firefox when running on Apple iPad. Firefox on desktop looks just fine.

### Steps To Reproduce

Firefox on Apple iPad only.

### Expected behaviour

Proper energy flows.

### Card Version

All versions up to current 6.6.4

### Home Assistant Version

2025.2.4

### Configuration

```shell

```

### Relevant log output

```shell

```

slipx06, 2025-02-17:
Those rendering issues usually mean a problem with the browser version. and legacy builds.  Have a look here https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v5.0.0
Unfortunately not much I can do

makcorner, 2025-02-17:
Thanks for your answer. This is not related to legacy builds. This is the 10th gen iPad (second to the newest available, still in sale), with current operating system and Firefox. Confirmed on two different tablets. It seems like iPad Firefox, even if current, cannot render power flow card correctly. Safari and Chrome - both on the same iPad - have no issues at all. Best would be you could check it yourself on some iPad, unless is there any way I can help?

slipx06, 2025-02-17:
@makcorner  yes I can reproduce this. 
If you click the hamburger menu and select  "request mobile site" it renders fine (and still shows the desktop version as well) It might be something to so with the user agent that Firefox is using but this is outside the scope of the card

makcorner, 2025-02-17:
Yes, after requesting mobile version it renders just fine - thanks. Anyway it is something strange, clearly related to Firefox on this particular platform.

closed 2025-02-17 by slipx06