slipx06/sunsynk-power-flow-card #635: 5/6 Essential Load in compact mode

roumano, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Actually in compact mode, we can see 4 Essential Load 

### Expected behaviour

- The documentation is wrong (it's write 3 `Essential Load `) : 

| Attribute | Requirement | Default | Description |
| --- | --- | --- | --- |
| additional_loads: | Optional | 0 | Display additional loads on the essential side (0/1/2/3/4/5/6). Three loads will only be visable when using the lite and compact card. Up to six loads can be displayed if using the wide screen display mode. Four or more loads are only visable on the full card if show_aux: false.|

- If possible, i would like to see 5 (or 6)
for me, it's sufficient space to add it : 

[image]

### Mode

lite

### Context / Reason

In smartphone, in the `full` mode, text size is too small for me,
and i like the actually rendering mode of compact mode
i would like to see more essential load

slipx06, 2025-02-20:
It wont work if you have 4 PV strings so I have to build in a a bunch of conditional checks.

rafalw1277, 2025-03-06:
I am also interested, 5/6 Essential Load in compact mode.
Has the problem been solved?
I only have 2 strings.

roumano, 2025-03-06:
> I am also interested, 5/6 Essential Load in compact mode. Has the problem been solved? I only have 2 strings.

no, it's still not possible (yet), but it's not a problem, it's a feature request

commit ad2cc5b7 (2025-04-05): feat: added new inverter models for EASun, Ferroamp, Sigenergy and Linky closes #657, #579
feat: six essential loads can now be displayed on the card without the need for wide screen display mode if certain conditions are met i.e. mppt <=3 on lite card and !show_aux on full card etc closes #635
closed 2025-04-05 by slipx06