slipx06/sunsynk-power-flow-card #641: Font Size Battery Capacity

Mights001, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hello guys,

When you get a chance, could you please add an option to adjust the font size of the current battery capacity display?

Thank you!

### Expected behaviour

[image]

### Mode

lite

### Context / Reason

 Thank you!

Ben00793, 2025-02-24:
Install card mod from HACS, if you haven't got it already.
Then paste this code at the end of your card's yaml code.
Then you can play around with the numbers. All the texts are grouped so changing one font size on this list will alter multiple texts on the card, but it's possible to find a sweet spot. I think st3 will alter the battery capacity.

```
card_mod:
  style: |

    .st3 {
      font-size: 10px !important;
    }
    .st4 {
      font-size: 15px !important;
    }
    .st10 {
      font-size: 18px !important;
    }
    .st14 {
      font-size: 15px !important;
    }
    .st13 {
      font-size: 20px !important;
    }

```

Mights001, 2025-02-25:
Hi, you are right, under st3 is this text, but all other too, and i want to change only this Text because this is to small for me. 

If i set this to 15px than it looks like this
[image]

martinisonline, 2025-03-01:
I would also appreciate to have this option too.
I have a relative small screen on my desk with this power-flow always running, and it's hard to spot the remaining battery capacity :).

Thank you.

slipx06, 2025-03-01:
I can assign it its own style and the you can use card mod to set the size to your preference

martinisonline, 2025-03-01:
> I can assign it its own style and the you can use card mod to set the size to your preference

If that is possible, could be a solution.
Thank you.

slipx06, 2025-03-03:
You can change the font by adjusting the following style:

```
card_mod:
  style: |

    .remaining-energy {
      font-size: 10px !important;
    }
```

martinisonline, 2025-03-04:
Excelent. Working flawless.

Many thanks.

Mights001, 2025-03-04:
Thank you so much!

commit 99f3f890 (2025-03-03): chore: bump version  closes #641