slipx06/sunsynk-power-flow-card #642: Wrong Unit of measurement in load area

DJTech84, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hi,
In the loads section I have the unit of measurement expressed in kw instead of W even though the sensor measures correctly.
(I have 28W not 28KW)
I managed to remove only the decimals but not to correct the unit of measurement, I think it is a possible bug? However, I saw that in the documentation with the possible configurations on the official page this value is shown correctly
I didn't find any sections describing how to fix the problem

[image]

### Expected behaviour

The Load will show 28W and not 28KW

### Card Version

6.6.5

### Home Assistant Version

core 2025.2.5  Supervisor 025.02.3

### Configuration

```shell

```

### Relevant log output

```shell

```

slipx06, 2025-02-25:
Hi. How is that sensor configured. Do you have a unit of measurement defined? You can check in the developer tools under states

DJTech84, 2025-02-25:
Hi
I see this configuration
state_class: measurement
unit_of_measurement: W
device_class: power
friendly_name: Carichi APP_B Potenza

slipx06, 2025-02-25:
Post a screenshot please e.g. as well as your config

[image]

DJTech84, 2025-02-26:
Hi, sorry for late reply.
[image]
[image]

DJTech84, 2025-02-26:
I see high value but the unit of measure is W
in this case physically I have 283 W but I see 283KW,
if disable autoscale the value is 283900W

[image]

slipx06, 2025-02-27:
Something odd with that sensor. When auto scale is disabled the card only converts the sensor state to a number and displays this together with the unit of measurement. I cant see what the issue is

DJTech84, 2025-02-27:
At the moment I opened a topic to understand if I was getting a parameter wrong or if it was a bug, I can try to create a virtual sensor that performs the right calculation for the card and I will update the result of the problem if it is solved.

DJTech84, 2025-03-01:
I removed the physical sensor and reconnected to tuya, now it recognizes the unit of measurement perfectly. I don't know what happened because that was configured 1 year ago and I haven't touched it anymore, so I only took the parameter requested by the card, however I think they are the latest updates of home assistant, recently they disabled a display that previously showed the data.
You can consider the problem solved! Thank you very much for the support and the tests we did together to try to see where the problem was.

closed 2025-03-01 by slipx06